astuff/avt_vimba_camera #1: more PixelFormat support

jmoreau-hds, 2020-01-29:
Hi,
I will use a Mako G192C camera, but the raw pixelformat is BayerGB8 (BayerGB10 also available).
In avt_vima_camera.cpp, I can read that implemented pixelformats are in:
static const char* PixelFormatMode[] = {
  "Mono8",
  "Mono12",
  "Mono12Packed",
  "BayerRG8",
  "BayerRG12Packed",
  "BayerRG12",
  "RGB8Packed",
  "BGR8Packed"};
Is it possible to add other bayer formats?
Or, how could I do it myself?
Thanks

jmoreau-hds, 2020-02-03:
Problem solved in my fork. Please consider to update this repo.
Bye

icolwell-as, 2020-02-06:
@jmoreau-hds Thanks for raising the issue! please open a pull request into this repo from your fork and I'll be happy to review and merge.

PR #3 (2020-03-06): Add More Pixel Formats
Cleaned up copy of #2 

Thanks @jmoreau-hds

Resolves #1
closed 2020-03-06 by icolwell-as