astuff/avt_vimba_camera #4: camera Ros synchronization

edob95, 2020-02-19:
Hi, 
I'm actually using a Mako camera and I should synchronize it with ros time in order to get images having their timestamp at acquisition time. For the moment I have found a way to get the camera time at acquisition instant instead of reception one but now I would like to give to the camera the correct ros time. 

A first idea I have is to set the camera timer to 0 when it is initialized saving also the ros time. Then  when a frame is acquired give to it the timestamp obtained as sum of the camera time at acquisition instant and the previously saved  ros time. 
Is there a better way to perform such operation? 

Thanks in advance and sorry if this is not the right place to ask a question.

icolwell-as, 2020-02-26:
Hi @edob95, this is actually a common problem with many sensor drivers (ROS or not). The root of the issue is that each sensor (and the main computer) typically has it's own on-board clock. Each of these clocks run at their own imperfect frequencies and would need to be constantly managed/synced so that they don't drift apart from each other. This is the fundamental problem that [PTP](https://en.wikipedia.org/wiki/Precision_Time_Protocol) was invented to solve.

So, back to ROS drivers, there are basically 3 options:
- Sync the sensor clock with the ROS master clock using PTP or hardware synchronization (GPS PPS, etc.). ROS uses the system time, so if system time is synced, ROS timestamps are valid.
- Re-Implement the algorithms behind the PTP protocol inside each ROS driver. Not a bad solution, but sort of re-inventing the wheel.
- Stamp the sensor data when it arrives at the computer using ROS time. This is currently what most ROS sensor drivers (at least the ones that I've seen) do since it is the simplest. Obviously the timestamp is not the acquisition time, but most people seem to find it sufficient for their purposes.

To give you an example of how complicated system-wide time synchronisation is, Baidu's Apollo team built their own custom hardware to manage camera triggering and time sync:
[Apollo ASU](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/Apollo_Sensor_Unit/Apollo_Sensor_Unit_Installation_Guide.md)

Hope this helps!

edob95, 2020-02-27:
Hi @icolwell-as,
Thank you very much for your answer. Actually I have discovered that the camera model that my laboratory bought (Mako G-192C) is one of the few not supporting PTP.
Consequently I have created a "synchronization protocol" of my own reinitializing the camera time at a known Ros Time and adding this quantity to the acquisition time of the image given by the camera and properly converted.
I will try to estimate somehow the delay between the "real" acquisition time and the image timestamp. Anyway for the moment I don't see any other solution.

icolwell-as, 2020-03-03:
No problem! Glad you have sorted out a solution. If you end up deciding to upgrade your camera hardware, we sell the [Mako G-319C](https://autonomoustuff.com/product/allied-vision-mako-g-319c/) which has PTP support.

closed 2020-03-03 by icolwell-as