astuff/avt_vimba_camera #5: munmap_chunk(): invalid pointer on api_.start() in mono_camera.cpp

rcywongaa, 2020-03-12:
The same thing happens when using the newer vimba API, where the same error is thrown on
```
m_system.Startup();
```
Where `m_system` is initialized as follows
```
    AVT::VmbAPI::VimbaSystem &       m_system(AVT::VmbAPI::VimbaSystem::GetInstance()) ;                   // A reference to our Vimba singleton
    m_system.Startup();
```

icolwell-as, 2020-03-30:
Thanks for reporting this, unfortunately we don't have any USB hardware to test/reproduce with. We only use the Ethernet mako cameras.

StephanEll, 2020-04-20:
I have the same problem. Did anyone find a solution?

rcywongaa, 2020-04-21:
@StephanEll I have contacted Allied Vision on this and have verified that it is an issue with their USB Transport Layer which they'll have to fix on their side.  It appears they are aware of issues with using their cameras with ROS.  Unfortunately, it doesn't seem high on their priority list and I haven't heard from them since.

mrkschnk, 2020-04-22:
@StephanEll , @rcywongaa Allied Vision is very interested in enabling the ROS community to use Vimba with ROS. But since we are not the experts in ROS, we need you to find the problem.
There may be a conflict with external libraries when using the USB interface with ROS. To further narrow this down, it would be great if you could communicate your used system configuration (like Host system / architecture, version from OS, Vimba, ROS, libboost, OpenCV) to the Allied vision support.

rcywongaa, 2020-04-23:
@mrkschnk That's good to know.

I've spoken with one of Allied Vision's application engineer (Jan Lemke).  He was able to reproduce the issue on his side and has confirmed that it was an issue with the transport layer provided by the vimba library.  I believe he has reported to the relevant product manager already.
Below is a sample ROS project that exhibits the problem (which I've also sent to Jan).
https://github.com/rcywongaa/vimba_invalid_pointer

mrkschnk, 2020-08-20:
@StephanEll , @rcywongaa Allied Vision released a new Vimba version 4.1 to address your experienced problem. Happy to get your feedback on this: https://www.alliedvision.com/en/products/software.html

icolwell-as, 2021-11-15:
Closing since the issue appears to be resolved.

closed 2021-11-15 by icolwell-as