astuff/avt_vimba_camera #18: when I run the "roslaunch avt_vimba_camera mono_camera.launch ",I met the problem

philtell, 2020-06-08:
tsinghualion@tsinghualion-Nuvo-6108GC:~/catkin_ws$ roslaunch avt_vimba_camera mono_camera.launch 
... logging to /home/tsinghualion/.ros/log/8eea1734-a937-11ea-85dc-78d00426c89d/roslaunch-tsinghualion-Nuvo-6108GC-6465.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tsinghualion-Nuvo-6108GC:45533/

SUMMARY
========

PARAMETERS
 * /camera/acquisition_mode: Continuous
 * /camera/acquisition_rate: 10
 * /camera/binning_x: 1
 * /camera/binning_y: 1
 * /camera/camera_info_url: file:///home/tsin...
 * /camera/decimation_x: 1
 * /camera/decimation_y: 1
 * /camera/exposure: 1000
 * /camera/exposure_auto: Continuous
 * /camera/frame_id: left_optical
 * /camera/gain: 0
 * /camera/gain_auto: Continuous
 * /camera/guid: 06-0536913354
 * /camera/height: 1440
 * /camera/ip: 192.168.1.210
 * /camera/pixel_format: BayerRG8
 * /camera/show_debug_prints: True
 * /camera/stream_bytes_per_second: 45000000
 * /camera/trig_timestamp_topic: 
 * /camera/trigger_source: FixedRate
 * /camera/whitebalance_auto: Continuous
 * /camera/width: 1920
 * /camera/x_offset: 0
 * /camera/y_offset: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    camera (avt_vimba_camera/mono_camera_node)

auto-starting new master
process[master]: started with pid [6477]
ROS_MASTER_URI=http://tsinghualion-Nuvo-6108GC:11311/

setting /run_id to 8eea1734-a937-11ea-85dc-78d00426c89d
process[rosout-1]: started with pid [6490]
started core service [/rosout]
process[camera-2]: started with pid [6503]
[camera-2] process has died [pid 6503, exit code -11, cmd /home/tsinghualion/catkin_ws/devel/lib/avt_vimba_camera/mono_camera_node __name:=camera __log:=/home/tsinghualion/.ros/log/8eea1734-a937-11ea-85dc-78d00426c89d/camera-2.log].
log file: /home/tsinghualion/.ros/log/8eea1734-a937-11ea-85dc-78d00426c89d/camera-2*.log

icolwell-as, 2020-06-08:
Hi @philtell, can you provide some additional details? Did you build the driver from source or are you running the driver installed from apt? Which version of the driver? which version of ROS? etc.

philtell, 2020-06-11:
the log file is:
[roslaunch][INFO] 2020-06-11 09:28:32,161: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2020-06-11 09:28:32,163: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2020-06-11 09:28:32,163: roslaunch starting with args ['/opt/ros/kinetic/bin/roslaunch', 'avt_vimba_camera', 'mono_camera.launch']
[roslaunch][INFO] 2020-06-11 09:28:32,163: roslaunch env is {'XDG_SESSION_TYPE': 'x11', 'LC_PAPER': 'en_HK.UTF-8', 'SHELL': '/bin/bash', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/share/gnome:/usr/local/share:/usr/share:/var/lib/snapd/desktop', 'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'CLUTTER_IM_MODULE': 'xim', 'JOB': 'dbus', 'SESSION': 'ubuntu', 'LIBGL_ALWAYS_SOFTWARE': '1', 'XMODIFIERS': '@im=fcitx', 'GTK2_MODULES': 'overlay-scrollbar', 'XDG_RUNTIME_DIR': '/run/user/1000', 'LC_ADDRESS': 'en_HK.UTF-8', 'PYTHONPATH': '/home/lzw/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/kinetic/lib/python2.7/dist-packages', 'ROS_ROOT': '/opt/ros/kinetic/share/ros', 'XDG_SESSION_ID': 'c1', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-XFO6cbYVIn', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'DESKTOP_SESSION': 'ubuntu', 'GTK_MODULES': 'gail:atk-bridge:unity-gtk-module', 'INSTANCE': '', 'LC_NAME': 'en_HK.UTF-8', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'LC_NUMERIC': 'en_HK.UTF-8', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'GENICAM_GENTL32_PATH': ':/opt/Vimba_4_0/VimbaGigETL/CTI/x86_32bit:/opt/Vimba_4_0/VimbaUSBTL/CTI/x86_32bit', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'USER': 'lzw', 'XDG_VTNR': '7', 'CMAKE_PREFIX_PATH': '/home/lzw/catkin_ws/devel:/opt/ros/kinetic', 'XAUTHORITY': '/home/lzw/.Xauthority', 'LANGUAGE': 'en', 'SESSION_MANAGER': 'local/lzw:@/tmp/.ICE-unix/1617,unix/lzw:/tmp/.ICE-unix/1617', 'SHLVL': '1', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'DISPLAY': ':0', 'QT_ACCESSIBILITY': '1', 'WINDOWID': '90177546', 'GPG_AGENT_INFO': '/home/lzw/.gnupg/S.gpg-agent:0:1', 'XDG_SESSION_DESKTOP': 'ubuntu', 'GDMSESSION': 'ubuntu', 'PKG_CONFIG_PATH': '/home/lzw/catkin_ws/devel/lib/pkgconfig:/opt/ros/kinetic/lib/pkgconfig:/opt/ros/kinetic/lib/x86_64-linux-gnu/pkgconfig', 'ROS_PACKAGE_PATH': '/home/lzw/catkin_ws/src:/opt/ros/kinetic/share', '_': '/opt/ros/kinetic/bin/roslaunch', 'GTK_IM_MODULE': 'fcitx', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg', 'ROS_MASTER_URI': 'http://localhost:11311', 'LC_TIME': 'en_HK.UTF-8', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/lzw', 'QT4_IM_MODULE': 'fcitx', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'HOME': '/home/lzw', 'LD_LIBRARY_PATH': '/home/lzw/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/usr/local/cuda-9.0/lib64:/usr/local/cuda-9.0/lib64', 'LANG': 'en_US.UTF-8', 'ROS_LOG_FILENAME': '/home/lzw/.ros/log/dc7b9ce6-ab82-11ea-aff1-9c7bef1ee058/roslaunch-lzw-9422.log', 'LC_MONETARY': 'en_HK.UTF-8', 'QT_LINUX_ACCESSIBILITY_ALWAYS_ON': '1', 'ROS_PYTHON_VERSION': '2', 'LC_IDENTIFICATION': 'en_HK.UTF-8', 'CUDA_HOME': '/usr/local/cuda-9.0', 'VTE_VERSION': '4205', 'XDG_CURRENT_DESKTOP': 'Unity', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'GNOME_KEYRING_PID': '', 'PAPERSIZE': 'a4', 'CYUSB_ROOT': '/home/lzw/\xe6\xa1\x8c\xe9\x9d\xa2/02\xe8\xb5\xb5\xe8\x80\x81\xe5\xb8\x88\xe7\xa8\x8b\xe5\xba\x8f/object_detection/camera_driver/cyusb_linux_1.0.4', 'ROS_ETC_DIR': '/opt/ros/kinetic/etc/ros', 'ROS_DISTRO': 'kinetic', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/lzw/catkin_ws/devel/share/common-lisp', 'QT_IM_MODULE': 'fcitx', 'LOGNAME': 'lzw', 'XDG_SEAT': 'seat0', 'GNOME_KEYRING_CONTROL': '', 'PATH': '/opt/ros/kinetic/bin:/home/lzw/anaconda3/bin:/home/lzw/bin:/home/lzw/.local/bin:/usr/local/cuda/bin:/usr/local/cuda-9.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/cuda-9.0/bin', 'TERM': 'xterm-256color', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'COMPIZ_CONFIG_PROFILE': 'ubuntu', 'SESSIONTYPE': 'gnome-session', 'IM_CONFIG_PHASE':  [... cut ...]
[roslaunch][INFO] 2020-06-11 09:29:17,858: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 139785570490112)>
[roslaunch.pmon][INFO] 2020-06-11 09:29:17,858: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139785570490112)>
[roslaunch][INFO] 2020-06-11 09:29:17,858: ... shutting down processing monitor complete
[roslaunch][INFO] 2020-06-11 09:29:17,858: done
[roslaunch.pmon][INFO] 2020-06-11 09:29:17,859: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139785570490112)>
[rospy.core][INFO] 2020-06-11 09:29:17,859: signal_shutdown [atexit]

philtell, 2020-06-11:
my Environment is Ubuntu16.04TLS , the ROS Version is Kinetic, The Python Version is Anocanda 3.7.6 and I have installed the Vimba SDK following the instructions of installation, The VimbaViewer can execute correctly and It can see the image of item. But when I run the command "roslaunch avt_vimba_comera mono_camera.launch", crash on the Terminal.
the log file show that

 "[roslaunch][ERROR] 2020-06-11 09:28:33,577: [camera-2] process has died [pid 9457, exit code -11, cmd /home/lzw/catkin_ws/devel/lib/avt_vimba_camera/mono_camera_node __name:=camera __log:=/home/lzw/.ros/log/dc7b9ce6-ab82-11ea-aff1-9c7bef1ee058/camera-2.log].
log file: /home/lzw/.ros/log/dc7b9ce6-ab82-11ea-aff1-9c7bef1ee058/camera-2*.log"

hoping your reply

philtell, 2020-06-11:
the Vimba Version is 4.0

icolwell-as, 2020-06-12:
Unfortunately nothing in those logs is giving me any clues.
You can also launch the node like this:
```
roslaunch avt_vimba_camera mono_camera.launch --screen
```
which should produce more verbose logs.

philtell, 2020-06-20:
thanks for your reply , I have solve the problem without any rectification. it's nice of you

icolwell-as, 2020-06-22:
Glad it's resolved!

closed 2020-06-22 by icolwell-as