astuff/avt_vimba_camera #20: why I only get the grayscale image?

yulan0215, 2020-07-13:
I have a camera: avt-camera GT6600C(Mono camera). When I ran avt_vimba_camera when I ran mono_camera_nodelet.launch in this path, it is grayscale image even if I ran ROSRUN_NAMESPACE=camera rosrun image_proc image_proc, it made no sense. Can u tell me what is the problem?

yulan0215, 2020-07-14:
I would like to ask one more question: I logged the launch file: mono_camera.launch and it showed the RGB image but the problem is that it can only show half view of the image. Can anyone give me the idea how to solve this problem? I used the camera is GT6600C anf it is the mono camera.
Thanks!

icolwell-as, 2020-07-14:
1. Where are you seeing the grayscale image? in RViz only, or is it also grayscale when you run `rosrun rqt_image_view  rqt_image_view`?
2. What version of ROS are you running?

> I would like to ask one more question: I logged the launch file: mono_camera.launch and it showed the RGB image but the problem is that it can only show half view of the image. Can anyone give me the idea how to solve this problem? I used the camera is GT6600C anf it is the mono camera.
> Thanks!
What do you mean you "logged" the image? saved it in a rosbag?

yulan0215, 2020-07-15:
> 1. Where are you seeing the grayscale image? in RViz only, or is it also grayscale when you run `rosrun rqt_image_view  rqt_image_view`?
> 2. What version of ROS are you running?
>
>
> What do you mean [...]

I ran the version ROS melodic and both Rviz and rqt_image_view I could not see the RGB. But when I ran the mono_camera.launch file it can only show half view but it is colorful image. Do you know what is the problem? Thank you very much!

yulan0215, 2020-07-15:
> I ran the version ROS melodic and both Rviz and rqt_image_view I could not see the RGB. But when I ran the mono_camera.launch file it can only show half view but it is colorful image. Do you know [...]

But when I launch the nodelet.launch file, it shows the full VoF, but it is grayscale image.

icolwell-as, 2020-07-17:
I recommend double-checking all the parameters in the launch file are appropriate for your camera.
I can't find a "GT6600C" Camera on the [AVT product page](https://www.alliedvision.com/en/products/cameras.html#spectrum%2F-1%2FmaxFrameRate%2F-1%2FresolutionCalc%2F-1%2Fcolor%2F-1%2FsensorSize%2F-1%2Fsensor%2F-1%2Fseries%2F35%2Ftext%2F%2Fintf%2F-1%2Finterf), but the 6400 has the following specs:
6480 (H) × 4860 (V) resolution
3.82 fps max frame rate
BayerRG8, BayerRG12, BayerRG12Packed formats.

Make sure the launch file matches those settings (but maybe pick a framerate of 1 Hz at first).
Also maybe try to set the x_offset and y_offset to zero, that's the only difference I can see between the "mono_camera.launch" and "mono_camera_nodelet.launch" files.

yulan0215, 2020-07-18:
> I recommend double-checking all the parameters in the launch file are appropriate for your camera.
> I can't find a "GT6600C" Camera on the [AVT product [...]

I modified some parameters but it made no sense, I did the calibration in ROS and I got the result: my camera was regarded as the narrow stereo camera, I guess that is the problem why I can only get half of the image view. I used the stereo camera launch file but the problem is: it could not get the camera. 
Could you please give me some ideas? Thank you very much~

yulan0215, 2020-07-20:
> I recommend double-checking all the parameters in the launch file are appropriate for your camera.
> I can't find a "GT6600C" Camera on the [AVT product [...]

I am wondering whether it is the problem of the frame id? I checked it is : <arg name="frame_id" default="left_optical" doc="The frame id of the camera"/>
In this case, I guess it may be the problem but I don not know how to modify....Could you give me some ideas? Thank you very much

icolwell-as, 2020-07-21:
> I used the stereo camera launch file but the problem is: it could not get the camera.

Do you have a single camera or two cameras in a stereo camera setup?
If you are just trying to get images from a single camera use the "mono_camera.launch" file.

yulan0215, 2020-07-21:
> > I used the stereo camera launch file but the problem is: it could not get the camera.
> 
> Do you have a single camera or two cameras in a stereo camera setup?
> If you are just trying to get images from a single camera use the "mono_camera.launch" file.
I have one camera, but the problem is that when I logged the camera via  "mono_camera.launch" file, it  can only showed the left side view, and when I did the calibration, it showed my camera is the narrow stereo camera, I don not know how to solve it , do you have any ideas?
I found that the frame id of the mono_camera.launch file is the left_optical, I don not know whether it is this problem?

yulan0215, 2020-07-21:
> > I used the stereo camera launch file but the problem is: it could not get the camera.
> 
> Do you have a single camera or two cameras in a stereo camera setup?
> If you are just trying to get images from a single camera use the "mono_camera.launch" file.
<launch>
  <arg name="name" default="camera" doc="The name of the camera"/>
  <arg name="ip" default="192.168.2.203" doc="The IP for the camera to connect to"/>
  <arg name="guid" default="50-0503343290" doc="The GUID for the camera to connect to"/>
  <arg name="frame_id" default="left_optical" doc="The frame id of the camera"/>
  <arg name="camera_info_url" default="file://$(find avt_vimba_camera)/calibrations/calibration_50-0503343290.yaml"/>

I am sorry that this is part of the mono_camera.launch file and I found that the   <arg name="frame_id" default="left_optical" doc="The frame id of the camera"/>.
I guess I need to have some modification of this part. Could you please tell me how to modify this part to get the full view of image? Thank you very much!

yulan0215, 2020-07-21:
> > I used the stereo camera launch file but the problem is: it could not get the camera.
> 
> Do you have a single camera or two cameras in a stereo camera setup?
> If you are just trying to get images from a single camera use the "mono_camera.launch" file.

When I launched the launch.file it showed: 
[ WARN] [1595356174.925484315]: The input topic '/camera/image_raw' is not yet advertised
[ WARN] [1595356174.928610185]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1595356178.312691631]: [/camera]: Feature PixelFormat is available now.
[ WARN] [1595356178.354878434]: [/camera]: Could not get feature IrisAutoTarget
 Error: Not found.
[ WARN] [1595356178.355536879]: [/camera]: Could not get feature IrisMode
 Error: Not found.
[ WARN] [1595356178.356593402]: [/camera]: Could not get feature IrisVideoLevelMax
 Error: Not found.
[ WARN] [1595356178.356977521]: [/camera]: Could not get feature IrisVideoLevelMin
 Error: Not found.
[ WARN] [1595356178.370544140]: Camera calibration file /opt/ros/melodic/share/avt_vimba_camera/calibrations/calibration_50-0503343290.yaml not found.

yulan0215, 2020-07-21:
> > I used the stereo camera launch file but the problem is: it could not get the camera.
> 
> Do you have a single camera or two cameras in a stereo camera setup?
> If you are just trying to get images from a single camera use the "mono_camera.launch" file.

The frame_id in this launch file is :left_optical(default).

icolwell-as, 2021-11-15:
Closing due to age. @yulan0215, please re-open if you are still having difficulty.

closed 2021-11-15 by icolwell-as