astuff/avt_vimba_camera #23: Issue of the published topic

yulan0215, 2020-09-18:
When I launched the launch.file, it showed:
[ WARN] [1600431979.605538713]: The input topic '/camera/image_raw' is not yet advertised
[ WARN] [1600431979.607691868]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1600431983.054640699]: [/camera]: Feature PixelFormat is available now.
[ WARN] [1600431983.097872189]: [/camera]: Could not get feature IrisAutoTarget
 Error: Not found.
[ WARN] [1600431983.098439489]: [/camera]: Could not get feature IrisMode
 Error: Not found.
[ WARN] [1600431983.099186270]: [/camera]: Could not get feature IrisVideoLevelMax
 Error: Not found.
[ WARN] [1600431983.099850811]: [/camera]: Could not get feature IrisVideoLevelMin
 Error: Not found.
I waant to know how to solve the problem of the topic: /camera/image_raw and /camera/camera_info

icolwell-as, 2020-09-18:
Please provide more information.

- Apart from a warning message, what is the expected result, and what is the result you actually got? Warnings alone don't always mean something is wrong or malfunctioning.
- What launch file did you use?
- What version of ROS are you using?
- What model of camera are you using? USB connection or Ethernet?

yulan0215, 2020-09-19:
> Please provide more information.
>
> * Apart from a warning message, what is the expected result, and what is the result you actually got? Warnings alone don't always mean something is wrong or [...]
Actually I used the launch file is :mono_camera.launch 
I used ROS melodic and my camera is GT6600 with ethernet connection
I hope to do camera calibration but the problem is those two topics are not advertised.
Besides, the biggest problem I encountered now is the size of the camera, every time when I launched the launch file:mono_camera.launch , it can only show the size:4096*4096, however, the original size of the image should be  4384*6576, I changed the size in the launch file:
<arg name="width" default="6576" doc="Width of the image (pixels)"/>
  <arg name="height" default="4384" doc="Height of the image (pixels)"/>
But it made no sense, the size of image is still 4096*4096.
Could you please tell me how to solve this problem? Thanks!

yulan0215, 2020-09-19:
Actually I used the launch file is :mono_camera.launch
I used ROS melodic and my camera is GT6600 with ethernet connection
I hope to do camera calibration but the problem is those two topics are not advertised.
Besides, the biggest problem I encountered now is the size of the camera, every time when I launched the launch file:mono_camera.launch , it can only show the size:4096*4096, however, the original size of the image should be 4384*6576, I changed the size in the launch file:
But it made no sense, the size of image is still 4096*4096. Could you please tell me how to solve this problem? Thanks!

icolwell-as, 2021-11-15:
Hi @yulan0215, in case you or someone else is still having this issue: The width and height of the image are capped at 4096 in the dynamic reconfigure configuration file. See `cfg/AvtVimbaCamera.cfg` to edit the max values for that parameter which should allow you to specify larger dimensions.

closed 2021-11-15 by icolwell-as