astuff/avt_vimba_camera #24: Gain cannot be set

michaelschleiss, 2020-10-02:
I cannot set the gain in dynamic reconfigure. I am getting the following warnings when the camera node is launched.

[ WARN] [1601656937.760431651]: [/cam_vo]: Could not get feature Gain
 Error: Not found.
[ WARN] [1601656937.764869844]: [/cam_vo]: Could not get feature DecimationHorizontal
 Error: Not found.
[ WARN] [1601656937.764939937]: [/cam_vo]: Could not get feature DecimationVertical
 Error: Not found.
[ WARN] [1601656937.770969267]: [/cam_vo]: Could not get feature PtpMode
 Error: Not found.
[ WARN] [1601656937.777396508]: [/cam_vo]: Could not get feature IrisAutoTarget
 Error: Not found.
[ WARN] [1601656937.777619918]: [/cam_vo]: Could not get feature IrisMode
 Error: Not found.
[ WARN] [1601656937.777709050]: [/cam_vo]: Could not get feature IrisVideoLevelMax
 Error: Not found.
[ WARN] [1601656937.777808284]: [/cam_vo]: Could not get feature IrisVideoLevelMin

Any idea about what this could be related.

icolwell-as, 2020-12-01:
What version of ROS do you have and what launch file are you using to start the driver?

icolwell-as, 2021-11-10:
Closing from lack of response.

closed 2021-11-10 by icolwell-as