astuff/avt_vimba_camera #26: Driver stamps images with ROS time rather than camera time

cparedis, 2020-11-29:
The frameCallback method stamps new images with rost_time = ros::Time::now();
We are synchronizing the clocks of all our cameras using PTP and it would therefore be of interest to stamp the images with the timestamp provided by the camera, which can be easily accessed as ((double)vimba_frame_ptr->GetTimeStamp())/((double)freq)

Seems like it could be a small fix to add a ROS parameter that would allow the user to select which clock to use (with the default being the ros_time for compatibility).

icolwell-as, 2020-12-01:
I agree, sounds like a great idea!

closed 2021-11-25 by icolwell-as

GarethZhang, 2022-01-31:
Hi @cparedis, 

Our team is also trying to set up PTP with Vimba cameras. After setting the `PtpMode=Slave` attribute in VimbaViewer and starting the PTP service, `PtpStatus` in VimbaViewer is stuck at `PtpStatus=Listening` as opposed to `PtpStatus=Slave`. Did you have a similar problem when working with PTP?

We are using host computer as master clock, without an external GNSS timing source, and want to set up Vimba cameras as slaves.

Thank you! Any suggestions are appreciated!

icolwell-as, 2022-01-31:
Hi @GarethZhang, I added a small section to the README regarding PTP sync:
https://github.com/astuff/avt_vimba_camera#clock-synchronization
In case you hadn't noticed it.
I was able to test PTP operation successfully when I tested some of the recent features added to this driver.
There could be a couple reasons the camera isn't using the computer as a master. What software are you using for the computer-based PTP master clock?

cparedis, 2022-02-03:
Haowei:
We did have a problem with PTP sync’ing that may be related.  We were using a Timemachines TM2000B as a master clock and could not get the cameras to sync.  After looking at the network traffic using wireshark, we found that by default the TM2000B was sending 128 sync messages per second.  The cameras (Allied Vision Mako G319-C) was expecting such sync messages at 1 per second.  When we reconfigured the master to broadcast sync messages at that frequency, everything worked fine.  I hope this helps.

Best regards,
CHRIS PAREDIS
Professor & BMW Endowed Chair in Automotive Systems Integration
College of Engineering, Computing and Applied Sciences<https://www.clemson.edu/cecas/>
Clemson University

4 Research Drive, Suite 330
Greenville, SC 29607
o 404-981-6549
a 864-283-7220 (Ms. Kay Faith)
***@***.******@***.***>

[signature_1757566267]<https://www.clemson.edu/>

GarethZhang, 2022-02-07:
Hi @cparedis ,

Thank you for the response! 

I've located that the issue was the Ethernet switch we were using. It somehow doesn't work properly with PTP. I'll keep an eye on this issue you mentioned when I engage the GNSS time source into the PTP in the near future. 

Best,
Haowei

PR #41 (2021-11-25): Use timestamp from image capture in ROS header
Resolves #26 by adding the ability to choose the timestamp from the image capture or use the time the image was received on the computer. Using the measurement time of the image is only really useful if PTP, or some other type of time sync setup, is enabled.

Tested with Mako G-319, didn't test with PTP enabled, but that only changes things on the camera side, not the driver.