astuff/avt_vimba_camera #27: ROS2 Version

JWhitleyWork, 2021-02-23:
I'm wondering if AutonomouStuff is interested in maintaining a ROS2 version of this driver. One has been created at https://github.com/neil-rti/avt_vimba_camera by @neil-rti but I'm not sure that RTI wants to maintain it long-term. If RTI _does_ want to maintain the ROS2 version, we should clarify on this repo that RTI's should be used for ROS2 and vice versa. If not, a PR should be made from Neil's repo to this repo (@icolwell-as - could you make a `ros2` branch in case this is the desired outcome?).

Serafadam, 2021-11-16:
@icolwell-as I'd be willing to help with that, as I am working on getting my USB AVT camera to run on Jetson Xavier platform  (version by @neil-rti works only with GIGe), here is my fork https://github.com/Serafadam/avt_vimba_camera

icolwell-as, 2021-11-16:
Awesome, thanks @Serafadam! Currently my plan is to clean up the ROS1 driver and add any outstanding feature requests. Then I intend to port the updated driver to ROS2 using your and Neil's repos as guides/examples. I'm hoping to mostly finish the ROS1 work this week. I'll make sure to get your help testing the ROS2 port, to ensure it works on your camera.

On that subject, have you had any issues using this driver or the existing ROS2 ports with your USB camera? Others have had difficulties in the past with USB it seems (https://github.com/astuff/avt_vimba_camera/issues/29). I only use ethernet cameras, so I can't reproduce.

icolwell-as, 2021-11-26:
Hi @Serafadam, just curious if you had any luck using this driver with your USB camera. I hope to finish up a new ROS1 release soon, I'd be happy to include any fixes that make it work better on USB cameras.

Serafadam, 2021-11-29:
Hi, I've been trying to run the camera, but so far no luck, I added some additional parameters, and while it seems that the camera is detected, and the node doesn't crash, no images are published by the driver.

icolwell-as, 2021-11-29:
Thanks @Serafadam! I reviewed the code and looked over some of the vimba documentation and still can't see anything obviously wrong.
Are you able to get image data when using the "Vimba Viewer" application that comes with the Vimba SDK?
Also, if you add a `ROS_INFO` print at the start of this function:
https://github.com/astuff/avt_vimba_camera/blob/master/src/mono_camera.cpp#L73
Does it ever print? I'm wondering if the Vimba API never calls the callback, or if something else on the ROS side is going on.

icolwell-as, 2021-11-29:
One other suggestion is to try the latest master branch from this repo, I've updated the VimbaCPP version that the driver uses and made other improvements, might be worth a try to see if anything changes.

Souphis, 2021-11-29:
Hi, I am working with @Serafadam  on this ROS2 port. I am also unable to start the stream from the camera. I think that ROS2 version should be written from scratch because it is really difficult to navigate in this current ROS2 version.

icolwell-as, 2021-11-29:
Hi @Souphis, thanks for sharing! I agree that the code is a bit unorganized, that's one of the reasons I haven't actually got started on the ROS2 port yet (no ROS2 driver exists in this repo), since I am fixing up the ROS1 version first, then porting that to ROS2. 

Have you had luck getting your USB cameras to work with Vimba Viewer?

Are you able to try the ROS1 driver? or are you only interested in ROS2? It would be helpful to know if the existing master branch (ROS1) of this repo works with USB cameras, and ideally fix that first before porting it to ROS2.

Souphis, 2021-11-29:
The camera is detected in the Vimba Viewer, however, I cannot start the acquisition (I think that it will work after proper configuration). I haven't tried to run USB cam with ROS1 driver, I will definitely do this as soon as possible.

@Serafadam maybe we should help in refactoring ROS1 driver and then migrate this to ROS2? This ROS2 driver is really difficult to debug.

Souphis, 2021-11-29:
@icolwell-as maybe I will try to migrate your version of ROS1 driver to ROS2 even it is not ready? It could speed up development.

icolwell-as, 2021-11-29:
Hi @Souphis, thanks for offering to try the ROS1 driver!
This is actually the last ROS1 PR left: https://github.com/astuff/avt_vimba_camera/pull/52
Once that merges, I plan to start on the ROS2 port immediately (unless we need to fix something related to USB cameras).

I always suggest trying Vimba Viewer first because it is 100% AVT code, so if you have any issues getting your camera to work with it, then the best thing to do is reach out to AVT support to sort out the cause of the problem.
Also, if it works with Vimba Viewer, then we know there isn't some existing hardware or environment/installation issue, so we can focus debug effort on the ROS driver code.

If you can get it working with Vimba Viewer, then the next step would be playing around with this ROS1 driver (which has been updated a lot since the original ROS2 fork from neil-rti) to better understand where something is going wrong. The Vimba SDK also has a lot of examples, so maybe compiling the examples and trying to get them to work with your camera is a good idea too, just to see what the driver is doing differently than the examples.

There's either some little detail the driver is missing, or some bug in the VimbaSDK that is causing these USB camera issues. Best approach is to sort out where the issue is first, then we can worry about the ROS2 port later.

Thanks for any time you are able to lend to help sort this out! I really appreciate it.

Souphis, 2021-12-01:
@icolwell-as Update from USB camera debugging:
* Camera is detected via Vimba Viewere. I have tested software and hardware trigger, everything works great.
* Camera is not working with current master of ROS1 driver. It looks like the problem lies in starting the acquisition, the console prints following: `[ERROR] [1638381465.331665831]: Could not start continuous image acquisition. 
 Error: Bad parameter.`. I suppose that the problem is with FrameObserver, however, I didn't investigate it.

icolwell-as, 2021-12-01:
Thanks @Souphis! Interesting, I'm not sure why that would be happening.
What model of camera do you have?

Souphis, 2021-12-01:
I am using this camera "Allied Vision 1800 U-1240c". I will try to debug this problem this week.

Souphis, 2021-12-01:
@icolwell-as  Ok, probably I found the source of the problem.
```  
if (trigger_source_int == Freerun || trigger_source_int == FixedRate || trigger_source_int == SyncIn1 ||
      trigger_source_int == Action0 || trigger_source_int == Action1)
  {
    // Create a frame observer for this camera
    SP_SET(frame_obs_ptr_,
           new FrameObserver(vimba_camera_ptr_,
                             std::bind(&avt_vimba_camera::AvtVimbaCamera::frameCallback, this, std::placeholders::_1)));
    camera_state_ = IDLE;
  }
```
In USB cameras there are following trigger sources: "Software", "Line0", "Line1", "Line2" and "Line3". I will fix this code.

icolwell-as, 2021-12-02:
Interesting! wouldn't you have seen the following error though?
```
ROS_ERROR_STREAM("Trigger mode " << TriggerMode[trigger_source_int] << " not implemented.");
```

Souphis, 2021-12-06:
@icolwell-as good news! I am finally managed to run a USB camera with ur driver. The camera is triggered by Line0. I've done this by adding this:
```
  std::string d;                                                                                                                       
  configureFeature("LineSelector", std::string{"Line0"}, d);                                                                           
  configureFeature("LineMode", std::string{"Input"}, d);           
  configureFeature("TriggerSource", std::string{"Line0"}, d);                                                                          
  configureFeature("TriggerMode", std::string{"On"}, d);
```

icolwell-as, 2021-12-06:
@Souphis that's great news! I'm glad it's working for you.
We can add the the "LineSelector" and "LineMode" to the .cfg file, so no code modifications are needed in the future. Feel free to open a PR if you wish, if not, I can get to it sometime next week. 
Thanks for sharing your findings!

Souphis, 2021-12-07:
@icolwell-as I am already implementing these changes, so I'll open PR this week.

Souphis, 2021-12-08:
@icolwell-as PR is ready. Also, I think that ROS2 version should be wrapped into a lifecycle node. Lifecycle node could enable easy procedure for features that cannot be changed during runtime.

Souphis, 2021-12-09:
Wow, nice work with this ROS2 version. If you don't mind, I would like to take part in the review of the ROS2 version ;)

icolwell-as, 2021-12-09:
@Souphis, yes please! happy to have you review. I will put up the first PR shortly. We are hoping to release the first ROS2 version very soon (Friday or Monday).

PR #56 (2021-12-10): ROS2 Port
This PR Resolves #27 by porting to ROS2.
I plan to update the README in another PR, and also add the USB changes from #54 in a separate PR.

Major Changes:
- There is no concept of dynamic reconfigure in ROS2, so a new approach to configuring the camera was created. The driver now queries the camera for all features and exposes them as parameters. This should make it easier to use any type of AVT camera in the future.
- Sync node was removed, this will also be removed from the ROS1 branch.
PR #63 (2021-12-13): 2001.1.0 Release
Resolves #27 by Releasing the first ROS2 version of this driver.

This 2001.1.0 version will coincide with an updated 1.1.0 release of the ROS1 driver.
closed 2021-12-13 by icolwell-as