astuff/avt_vimba_camera #28: Resolution is wrong when using binning / decimation.

MaxandreOgeret, 2021-02-25:
First of all, what is the difference between binning and decimation ?

I believe `decimation_x` and `binning_x` will reduce the width. And I believe `decimation_y` and `binning_y` will reduce the height. Am I right ? But 

When setting the launch parameters : 
 -  `binning_x` and `binning_y` to 2
 OR
 -  `decimation_x` and `decimation_y` to 2

I believe that the values of the parameters  `width` and `height` should be divided by `2` as well. Because those values will be outputted in the `camera_info` topic. am I right ?

It works fine for the y values (height) but doesn't for the x values (width). Setting the width to half will crop the image half just like if the decimation happened after the cropping which shouldn't be the case.

---

Config that works :
 -  `binning_x` and `binning_y` to 2 OR `decimation_x` and `decimation_y` to 2
- Height divided by 2 (`1032`)
- Width unchanged (`1544`)

In the topic `/camera_fx/image_raw` we get images of good resolution : `1032x772`

:warning: But the reported resolution in `/camera_fx/camera_info` is wrong, giving the values given as parameters : `1032x1544`.

---

Have you ever had this problem ?

Thanks !

icolwell-as, 2021-03-02:
Hi @MaxandreOgeret, I found a good description of binning and decimation on the LUCID camera website:
https://support.thinklucid.com/knowledgebase/binning-and-decimation-on-lucid-cameras/

Regarding our camera Model (Mako G-319), I also found the following flow chart in the [technical manual](https://cdn.alliedvision.com/fileadmin/content/documents/products/cameras/Mako/techman/Mako_TechMan_en.pdf).

[image]

According to that chart, it looks like ROI cropping happens before any decimation or binning. 
I also noticed in the bottom-left there it says "BinningY is supported by Mako G-040C, G-158C, and G-234C only", so it looks like binning isn't supported for you.

Regardless, the part of the code that I believe is responsible for the `camera_info` output is here:
https://github.com/astuff/avt_vimba_camera/blob/master/src/avt_vimba_camera.cpp#L1084

I reviewed the code but don't see anything out of the ordinary, or anything different between x and y.
At the bottom of that function it prints a bunch of information out to "ROS_INFO_STREAM". What does that output look like for you when you run the driver?

icolwell-as, 2021-03-02:
Another suggestion is to open vimba viewer and try modifying the settings directly on the camera and looking at the output. This removes the ROS driver from the equation and may give you a better feel for what settings result in what type of output from the camera.

MaxandreOgeret, 2021-03-04:
Thanks for your answer ! I have more insight now with the links you provided. And I will try the same configuration in Vimba viewer and update here.

PR #46 (2021-11-25): Parameter Refactor
Resolves #37 by reviewing all parameters and fixing/updating as-needed. 
I decided to remove the `roi_*` parameters in favour of the feature names the camera uses: `width, height, offset_x, offset_y`
I think this will make it much easier for people to configure the camera as they desire, since the parameter names match what they see in VimbaViewer. The roi details of of the published camera_info topic will be generated according to the vimba settings instead of the other way around.

I believe this also resolves #28 as there were some existing bugs where the camera_info topi [... cut]
closed 2021-11-25 by icolwell-as