astuff/avt_vimba_camera #29: USB camera, error "Could not start Vimba system: TL not loaded."

maciej-3, 2021-03-02:
Hello! I plugged my USB Alvium 1800 U-319c camera and installed this ROS driver on my ubuntu 20.04 with ROS Noetic according to the instruction `sudo apt install ros-$ROS_DISTRO-avt-vimba-camera`.

When I do `roslaunch avt_vimba_camera mono_camera.launch` on a sourced terminal, it says: `[ERROR] [1614724799.379538586]: [Vimba System]: Could not start Vimba system: TL not loaded.`

Then, I downloaded `Vimba 4.2` from https://www.alliedvision.com/en/products/software.html, installed it, and set up the proper path to `GENICAM_GENTL64_PATH` so that when I do `echo $GENICAM_GENTL64_PATH` it gives me the correct path. However, when I try `roslaunch avt_vimba_camera mono_camera.launch`, it gives me the same error. 

I can add that when I open the VimbaViewer from `/vimba/Vimba_4_2/Tools/Viewer/Bin/x86_64bit`, everything works - I can see the listed camera and the video.

Please help! How can successfully `roslaunch` your driver on ROS?

PhilippSchmaelzle, 2021-03-03:
Hi, I'm using a GigE camera. But I received the same error whenever I forgot to execute the installation of `lib/VimbaX_X/VimbaGigETL/Install.sh`
There seems to be the same Installation possibility for USB cameras.  `sudo ... lib/VimbaX_X/VimbaUSBTL/Install.sh`

Was the solution for me

maciej-3, 2021-03-03:
> Hi, I'm using a GigE camera. But I received the same error whenever I forgot to execute the installation of `lib/VimbaX_X/VimbaGigETL/Install.sh`
> There seems to be the same Installation possibility for USB cameras. `sudo ... lib/VimbaX_X/VimbaUSBTL/Install.sh`
> 
> Was the solution for me

Thx @PhilippSchmaelzle for your suggestion, but in my case, I do install `Vimba_4_2/VimbaUSBTL/Install.sh` and make sure that the `GENICAM_GENTL64_PATH` is properly set. Nevertheless, the error persists.

maciej-3, 2021-03-14:
I reported this issue to Allied VIsion, and indeed, they agreed (after 2-hour remote testing on my PC) that this is a Vimba-related issue. I am waiting for their fix then.

maciej-3, 2021-04-09:
Allied Vision said they have never bumped into this bug and they are working on it. In the meantime however, I have changed my PC and now `Vimba_4_2/VimbaCPP/Examples/ListCameras/Build/Make/binary/x86_64bit/ListCameras` works. However, this ROS driver still does not work with my USB camera as it keeps displaying this error `[Vimba System]: Could not start Vimba system: TL not loaded.`.

Does this ROS driver account for the 64bit version of vimba 4.2? I am asking because vimba 4.2 sets path to `GENICAM_GENTL64_PATH ` only. `GENICAM_GENTL32_PATH ` is empty. Even when I try to set the same path for `GENICAM_GENTL32_PATH ` as it is for `GENICAM_GENTL64_PATH `, this ROS driver still does not work.

Any ideas on how to fix that?

Mickeyyyang, 2021-04-27:
> Allied Vision said they have never bumped into this bug and they are working on it. In the meantime however, I have changed my PC and now [...]

Hi, Excuse me, have you solved this problem, I also encountered this problem...

maciej-3, 2021-04-27:
> Hi, Excuse me, have you solved this problem, I also encountered this problem...

Hello. I bypassed this problem by writing simple drivers in python based on `Vimba_4_2/VimbaPython/examples/`

Mickeyyyang, 2021-04-28:
Hi, Maciej~
     OK.
     Yes, I think you are right. I also made changes based on the official
Vimaba SDK, but it has not been successful. Can you share the driver file
that you have changed? Thx~
     Best wishes~

maciej-3, 2021-04-28:
> Hi, Maciej~ OK. Yes, I think you are right. I also made changes based on the official Vimaba SDK, but it has not been successful. Can you share the driver file that you have changed? Thx~ Best wishes

You can find my USB driver here: [https://github.com/maciej-3/pp_alvium_driver](https://github.com/maciej-3/pp_alvium_driver)

icolwell-as, 2021-11-26:
@maciej-3 and @Mickeyyyang if there is anything you have discovered that may help this driver work better on USB cameras, I'd be happy to accept a PR or create the PR myself. Hoping to do a new ROS1 release soon, so would be good to get the fixes into the new release.

Thanks!

closed 2021-11-26 by icolwell-as

icolwell-as, 2021-12-08:
For those who may come across this in the future, USB support has been added by this PR:
https://github.com/astuff/avt_vimba_camera/pull/54