astuff/avt_vimba_camera #31: Is there a way to get images' timestamps in the Unix time?

maciej-3, 2021-05-14:
Is there a way to get images' timestamps in the [Unix time]https://en.wikipedia.org/wiki/Unix_time)?

My USB camera seems to measure time (in python - `frame.get_timestamp()`) since plugged into the PC.

icolwell-as, 2021-05-20:
Hi @maciej-3, in general, ROS drivers will report messages in "ROS Time", which is the same as the system time of the computer running ROS. So if your computer's time is correct, then it should already be in Unix time.
See here for more details:
https://answers.ros.org/question/189867/what-is-the-timestamp/

maciej-3, 2021-05-22:
> ROS drivers will report messages in "ROS Time", which is the same as the system time of the computer running ROS.

Many thanks for your reply. The only thing is that the ROS time from this driver is not the exact time of image acquisition because it does not come from the camera itself. It is simply the time of the execution of the next line of code after acquiring an image from the camera:
`https://github.com/astuff/avt_vimba_camera/blob/master/src/mono_camera.cpp` lines `72` and `73`
```
void MonoCamera::frameCallback(const FramePtr&amp; vimba_frame_ptr) 
{  ros::Time ros_time = ros::Time::now();
```

This might be a bit of a problem for high-precision and dynamic tasks, especially when the camera must be synchronised with other sensors (in my case, it is a lidar).

What I meant in my first post was that if you do `frame.get_timestamp()`, it is not a Unix timestmap, but rather a timestamp measured from the moment when the camera was plugged into the PC.

Is there any way to get the precise Unix timestamp from the exact moment when the image was taken (and not from the time when the next line of code after the image acquisition was executed or the time measured from the moment when the camera was plugged into the PC)?

icolwell-as, 2021-05-25:
Hi @maciej-3, thanks for clarifying. I'm not super familiar with the Vimba API/library unfortunately, but it seems like the `get_timestamp()` function may be returning non-unix time because the camera hardware is not synchronized to a clock.

Based on this other [comment](https://github.com/astuff/avt_vimba_camera/issues/26), you may need to setup PTP sync first, then the `get_timestamp()` function may return the correct unix time of measurement. The driver was likely written to use ROS time instead since most people aren't using PTP with the camera.

icolwell-as, 2021-11-10:
Closing due to age, please re-open if there is still something to discuss.

closed 2021-11-10 by icolwell-as