astuff/avt_vimba_camera #47: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘offset_y’; did you mean ‘roi_offset_y’?

GarethZhang, 2021-11-25:
Hi, 

Thank you for the amazing driver first of all!

I'm encountering this following error when compiling (catkin_make) the latest GitHub version. I'm using Vimba_5.0 library in Noetic with Mako g-319c. I suppose these errors come from refactoring?

```
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:917:14: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_device_key’
  917 |   if (config.action_device_key != config_.action_device_key || on_init_) {
      |              ^~~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:917:43: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_device_key’
  917 |   if (config.action_device_key != config_.action_device_key || on_init_) {
      |                                           ^~~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:919:47: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_device_key’
  919 |     setFeatureValue("ActionDeviceKey", config.action_device_key);
      |                                               ^~~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:921:14: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_group_key’
  921 |   if (config.action_group_key != config_.action_group_key || on_init_) {
      |              ^~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:921:42: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_group_key’
  921 |   if (config.action_group_key != config_.action_group_key || on_init_) {
      |                                          ^~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:923:46: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_group_key’
  923 |     setFeatureValue("ActionGroupKey", config.action_group_key);
      |                                              ^~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:925:14: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_group_mask’
  925 |   if (config.action_group_mask != config_.action_group_mask || on_init_) {
      |              ^~~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:925:43: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_group_mask’
  925 |   if (config.action_group_mask != config_.action_group_mask || on_init_) {
      |                                           ^~~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:927:47: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘action_group_mask’
  927 |     setFeatureValue("ActionGroupMask", config.action_group_mask);
      |                                               ^~~~~~~~~~~~~~~~~
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp: In member function ‘void avt_vimba_camera::AvtVimbaCamera::updateGainConfig(avt_vimba_camera::AvtVimbaCamera::Config&)’:
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:1041:14: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘gain_auto_adjust_tol’; did you mean ‘gain_auto_tol’?
 1041 |   if (config.gain_auto_adjust_tol != config_.gain_auto_adjust_tol || on_init_) {
      |              ^~~~~~~~~~~~~~~~~~~~
      |              gain_auto_tol
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:1041:46: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘gain_auto_adjust_tol’; did you mean ‘gain_auto_tol’?
 1041 |   if (config.gain_auto_adjust_tol != config_.gain_auto_adjust_tol || on_init_) {
      |                                              ^~~~~~~~~~~~~~~~~~~~
      |                                              gain_auto_tol
/workspace/src/avt_vimba_camera/src/avt_vimba_camera.cpp:1044:52: error: ‘avt_vimba_camera::AvtVimbaCamera::Config’ {aka ‘class avt_vimba_camera::AvtVimbaCameraConfig’} has no member named ‘gain_auto_adjust_tol’; did you mean ‘gain_auto_tol’?
 1044 |                     static_cast<VmbInt64_t>(config.gain_auto_adjust_tol));
      |                                                    ^~~~~~~~~~~~~~~~~~~~
      |                                                    gain_auto_tol
```

Thank you!

icolwell-as, 2021-11-25:
Hi @GarethZhang 
Your guess is correct, this is due to the ongoing updates to the driver. 
The latest master branch should be ok now, you might need to do a `git pull` and `catkin clean` before rebuilding it. Also make sure you remove the apt-installed version (`sudo apt remove ros-$ROS_DISTRO-avt-vimba-camera`) in case you have that installed as well.

You may want to stick to version 0.0.12 since master will be changing a lot in the next couple of days. But feel free to use master if it compiles and works of course.

Hope this helps!

GarethZhang, 2021-11-25:
@icolwell-as Thank you for the quick response!

FYI, latest master branch works well :) I'll probably stick with this one for now.

closed 2021-11-25 by GarethZhang