astuff/avt_vimba_camera #49: Cannot get Iris related features

GarethZhang, 2021-11-25:
Hi,

When I run the driver using `G-319.launch`, following warnings related to Iris show up. I am compiling using `catkin_make` in ROS noetic using Vimba 5.0 library and G-319C camera. Compiling under 2c37ec8097e1e12ba5e8348f6090ec4df68251d9

```
[ WARN] [1637874022.257594157]: [/camera]: Could not get feature IrisAutoTarget
 Error: Not found.
[ WARN] [1637874022.257746718]: [/camera]: Could not get feature IrisMode
 Error: Not found.
[ WARN] [1637874022.257874293]: [/camera]: Could not get feature IrisVideoLevelMax
 Error: Not found.
[ WARN] [1637874022.258003698]: [/camera]: Could not get feature IrisVideoLevelMin
 Error: Not found.
```

Thank you! Any insights would be appreciated!

icolwell-as, 2021-11-25:
Hi @GarethZhang, that is perfectly normal, the G-319 doesn't have Iris-related features. You can just ignore these warnings.
I'm actually working on improving these types of errors so they are more helpful in the future.

closed 2021-11-25 by icolwell-as

GarethZhang, 2021-11-25:
Thank you! @icolwell-as