astuff/avt_vimba_camera #73: How to configure PTP with Mako G-319C

GarethZhang, 2022-01-31:
Hi,

This issue may not be related to the Vimba camera driver. 

Our team is hoping to enable PTP on our Vimba cameras, all of Mako G-319C model. We want to use our host computer as the PTP master clock and cameras as the slave. The cameras are connected to host computer via an Ethernet switch.

Following are our current procedures:
1. Set the `PtpMode` attribute in VimbaViewer to `Slave`
2. Start PTP service on our host computer.

Problems we are encountering:
The `PtpStatus` attribute in VimbaViewer is stuck at `Listening` which means it keeps looking for the best master clock on the network.

I'm curious to know if you have had any similar issues with setting up PTP on Vimba cameras?

What should be the expected output I am looking for to make sure PTP is properly synchronized.

Thank you!

icolwell-as, 2022-01-31:
Oops, I answered your other comment first. Anyway, what software are you using for the PTP master? Have you had luck syncing other devices as slaves to this master?

GarethZhang, 2022-01-31:
Hi @icolwell-as ,

Thanks for the quick response! I'm posting previous discussion here for future reference: https://github.com/astuff/avt_vimba_camera/issues/26#issuecomment-1026100961

I was using `linuxptp` and enabled PTP through VimbaViewer following the guide you provided in the README. To start PTP on Ubuntu 20.04, I followed [Section 16 PTP Quick Start Guide](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2.1.x.pdf) from Ouster manual and used `ptp4l` and `phc2sys` services.

Also, I was able to sync an Ouster LiDAR to my host computer via direction Ethernet connection. However with Vimba camera, I've added the switch (due to PoE) so I'm thinking this could potentially be a network problem.

If you don't mind, can you share your previous setup? e.g., were you trying to syncing with one camera to a host computer? What model of Ethernet switch were you using (I suspect switch needs to be PTP enabled)? What commands did you run to start PTP?

Thank you! Sorry for these many questions :(

icolwell-as, 2022-02-01:
Hi @GarethZhang, I think I've only used PTP with the Makos directly connected to the computer, but I've used PTP with other equipment via switches. There is nothing special about the switch since the PTP data is UDP, so generally PTP should work as long as other network traffic does too.

There is also this application note that may be useful:
https://cdn.alliedvision.com/fileadmin/content/documents/products/cameras/various/appnote/GigE/PTP_IEEE1588_with_Prosilica_GT_GC_Manta.pdf
Might be good to read that over and double-check your settings. However, if I remember correctly, all you need to do is set PtpMode to slave and watch for the PtpStatus.

I suggest using wireshark to debug PTP stuff, it has built-in support for parsing and displaying PTP packets, so it's fairly easy to see what's going on.

Finally, I also use these 2 commands to debug linuxptp:

```
sudo pmc -u -b 0 'GET CURRENT_DATA_SET'
sudo pmc -u -b 0 'GET TIME_STATUS_NP'
```
They will show you the status of PTP in the context of linuxptp. Might be useful. Also ptp4l can be run with varying levels of log output, that is also useful for seeing when connections happen, etc.

GarethZhang, 2022-02-07:
Hi @icolwell-as ,

Thank you so much for the response! They are really comprehensive!

I've located the issue and it is most likely to be the Ethernet switch we were using. We were trying to sync PTP with an Ouster and an AVT camera. The Ouster PTP stopped working when we connect it via the Ethernet switch, and thus we figured it's probably the switch that prevented AVT camera from properly syncing.

The application note and commands are really helpful! I'll try it out this week and hopefully it works after we replace our current Ethernet switch!

Thank you!

closed 2022-02-07 by GarethZhang

icolwell-as, 2022-02-07:
Thanks @GarethZhang for the update! It's always beneficial to hear the experience of others.

May I ask what model of switch you are using? Just in case I need to avoid it in the future.

I've just looked again, and apparently PTP uses multicast, which I didn't know or had forgotten.
Anyway, if your switch has some ability to configure it, I highly recommend disabling "IGMP Snooping", we've had issues in the past with IGMP snooping not working correctly (or our applications not playing nicely with IGNP snooping) when doing multicast stuff.

GarethZhang, 2022-02-07:
@icolwell-as We were using [Juniper EX4300](https://www.juniper.net/us/en/products/switches/ex-series/ex4300-enterprise-switch.html) and it didn't mention any PTP capabilities. btw We got around the Ouster PTP issue by connecting an extra Ethernet injector between Ouster and the switch. Hopefully we can do some trick similar to AVT camera.

Thanks for the note on IGMP! I'll take a look into this when I have got time.