astuff/avt_vimba_camera #79: No image data in the published topic

panc8170, 2022-02-25:
Hey there,

I am using ROS noetic and I deployed your drivers by extracting them and making them into package by catkin workspace. The drivers do run (I had to manually edit GUID in Roslaunch file) and they do publish a camera/image-raw topic but when we try to view it with rqt_image_view - we dont have any images.

The camera works fine in Vimba Viewer SDK run from terminal. So SDK install is fine.

Can you point us in right direction to get this working?

The sudo install version of this same package gives us error of not able to start continuous capturing of images.

icolwell-as, 2022-02-25:
What camera are you using? (model number) Is it USB or ethernet?
Do you see any ROS output (warnings or errors) on the command line when you run the driver? If you do, please copy-paste them here.

panc8170, 2022-02-26:
Camera Model : Allied Vision Alvium 1800 U-158c, 1/2.9" 1.58MP C-Mount, USB 3.1 Color Camera
Just to give you a context I installed Vimba USBTL using sudo install.sh as directed : [HERE](https://cdn.alliedvision.com/fileadmin/content/documents/products/software/software/Vimba/appnote/Vimba_installation_under_Linux.pdf)

Then I downloaded and copied your github files in ~catkin_ws/src. Catkin_make needed some changes in avt_vimba_camera.cpp - instead of line_selector I had to rename it to trigger_selector (otherwise the catkin_make would fail)

After catkin_make success, I edited the mono_camera.launch file for GUID : 01HLW my camera serial

ros launch file run using ROS command : roslaunch avt_vimba_camera mono_camera.launch
(I dont have the camera with me right now so it cannot connect to it, but when its plugged in camera does get connected)

##########################################################################
OUTPUT
##########################################################################
... logging to /home/tanmay/.ros/log/1cd2f822-9743-11ec-b5d8-ddfb3e6f0cbd/roslaunch-tanmay-VirtualBox-3447.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tanmay-VirtualBox:45347/

SUMMARY
========

PARAMETERS
 * /camera/acquisition_mode: Continuous
 * /camera/acquisition_rate: 10
 * /camera/balance_ratio_abs: 1.0
 * /camera/balance_ratio_selector: Red
 * /camera/binning_x: 1
 * /camera/binning_y: 1
 * /camera/camera_info_url: file:///home/tanm...
 * /camera/decimation_x: 1
 * /camera/decimation_y: 1
 * /camera/exposure: 50000
 * /camera/exposure_auto: Continuous
 * /camera/exposure_auto_alg: FitRange
 * /camera/exposure_auto_max: 50000
 * /camera/exposure_auto_min: 50000
 * /camera/exposure_auto_outliers: 50000
 * /camera/exposure_auto_rate: 50000
 * /camera/exposure_auto_target: 50000
 * /camera/exposure_auto_tol: 5
 * /camera/frame_id: camera
 * /camera/gain: 0
 * /camera/gain_auto: Continuous
 * /camera/gain_auto_adjust_tol: 5
 * /camera/gain_auto_max: 32
 * /camera/gain_auto_min: 0
 * /camera/gain_auto_outliers: 0
 * /camera/gain_auto_rate: 50
 * /camera/gain_auto_target: 50
 * /camera/guid: 01HLW
 * /camera/height: 1440
 * /camera/ip: 
 * /camera/iris_auto_target: 50
 * /camera/iris_mode: Continuous
 * /camera/iris_video_level_max: 110
 * /camera/iris_video_level_min: 110
 * /camera/line_mode: Input
 * /camera/line_selector: Line0
 * /camera/offset_x: 0
 * /camera/offset_y: 0
 * /camera/pixel_format: BayerRG8
 * /camera/print_all_features: False
 * /camera/ptp_mode: Off
 * /camera/ptp_offset: -37
 * /camera/stream_bytes_per_second: 45000000
 * /camera/sync_in_selector: SyncIn1
 * /camera/sync_out_polarity: Normal
 * /camera/sync_out_selector: SyncOut1
 * /camera/sync_out_source: GPO
 * /camera/trigger_activation: RisingEdge
 * /camera/trigger_delay: 0.0
 * /camera/trigger_mode: On
 * /camera/trigger_selector: FrameStart
 * /camera/trigger_source: FixedRate
 * /camera/use_measurement_time: False
 * /camera/whitebalance_auto: Continuous
 * /camera/whitebalance_auto_rate: 100
 * /camera/whitebalance_auto_tol: 5
 * /camera/width: 1920
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    camera (avt_vimba_camera/mono_camera_node)

auto-starting new master
process[master]: started with pid [3457]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1cd2f822-9743-11ec-b5d8-ddfb3e6f0cbd
process[rosout-1]: started with pid [3467]
started core service [/rosout]
process[camera-2]: started with pid [3470]
[ INFO] [1645907513.539762065]: [Vimba System]: AVT Vimba System initialized successfully
[ INFO] [1645907513.542873113]: Searching for cameras ...
[ INFO] [1645907516.356571465]: Trying to open camera by ID: 01HLW
### HERE ONWARDS THE CAMERA WILL GET CONNECTED AND THE LAST LINE WOULD BE STARTING CONTINOUS IMAGE ACQUISITION####

I can see camera/image and such published topics under rostopic list. BUT, when I open it using rqt_image_view and select those, NADA. Same with adding a subscriber node in the Rviz - no output.

I will post a final exact output when I go to place and connect the camera.

icolwell-as, 2022-02-28:
@panc8170, I'll wait to see your full output from the driver before digging too deep, but since you mentioned it's a USB camera: You are definitely using either the `ros1_master` branch or version `1.1.0` right? USB support was only added in 1.1.0 which is why I ask.

Ronak2904, 2022-03-01:
I am working with @panc8170 on the camera and to add to @panc8170  issue. Here is the output that we are receiving.

<pre>... logging to /home/autodriveii/.ros/log/b67b61e4-9987-11ec-860a-292f171fa728/roslaunch-autodriveii-X299X-AORUS-MASTER-2998.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

<b>started roslaunch server http://autodriveii-X299X-AORUS-MASTER:36209/</b>

SUMMARY
========

PARAMETERS
 * /camera/acquisition_mode: Continuous
 * /camera/acquisition_rate: 10
 * /camera/balance_ratio_abs: 1.0
 * /camera/balance_ratio_selector: Red
 * /camera/binning_x: 1
 * /camera/binning_y: 1
 * /camera/camera_info_url: file:///opt/ros/n...
 * /camera/decimation_x: 1
 * /camera/decimation_y: 1
 * /camera/exposure: 50000
 * /camera/exposure_auto: Continuous
 * /camera/exposure_auto_alg: FitRange
 * /camera/exposure_auto_max: 50000
 * /camera/exposure_auto_min: 50000
 * /camera/exposure_auto_outliers: 50000
 * /camera/exposure_auto_rate: 50000
 * /camera/exposure_auto_target: 50000
 * /camera/exposure_auto_tol: 5
 * /camera/frame_id: camera
 * /camera/gain: 0
 * /camera/gain_auto: Continuous
 * /camera/gain_auto_adjust_tol: 5
 * /camera/gain_auto_max: 32
 * /camera/gain_auto_min: 0
 * /camera/gain_auto_outliers: 0
 * /camera/gain_auto_rate: 50
 * /camera/gain_auto_target: 50
 * /camera/guid: 01HLW
 * /camera/height: 1440
 * /camera/ip: 
 * /camera/iris_auto_target: 50
 * /camera/iris_mode: Continuous
 * /camera/iris_video_level_max: 110
 * /camera/iris_video_level_min: 110
 * /camera/offset_x: 0
 * /camera/offset_y: 0
 * /camera/pixel_format: BayerRG8
 * /camera/print_all_features: False
 * /camera/ptp_mode: Off
 * /camera/ptp_offset: -37
 * /camera/stream_bytes_per_second: 45000000
 * /camera/sync_in_selector: SyncIn1
 * /camera/sync_out_polarity: Normal
 * /camera/sync_out_selector: SyncOut1
 * /camera/sync_out_source: GPO
 * /camera/trigger_activation: RisingEdge
 * /camera/trigger_delay: 0.0
 * /camera/trigger_mode: On
 * /camera/trigger_selector: FrameStart
 * /camera/trigger_source: FixedRate
 * /camera/use_measurement_time: False
 * /camera/whitebalance_auto: Continuous
 * /camera/whitebalance_auto_rate: 100
 * /camera/whitebalance_auto_tol: 5
 * /camera/width: 1920
 * /rosdistro: noetic
 * /rosversion: 1.15.14
[... 68 lines cut ...]
<font color="#C4A000">[ WARN] [1646156873.394073056]: Could not get feature PtpMode, your camera probably doesn&apos;t support it.</font>
[ INFO] [1646156873.394107011]: Updating PixelFormat config:
[ INFO] [1646156873.409140851]:  - PixelFormat set to BayerRG8
[ INFO] [1646156873.409217764]: Updating Acquisition and Trigger config:
[ INFO] [1646156873.410777048]:  - AcquisitionMode set to Continuous
<font color="#C4A000">[ WARN] [1646156873.410851318]: Could not get feature AcquisitionFrameRateAbs, your camera probably doesn&apos;t support it.</font>
[ INFO] [1646156873.412340223]:  - TriggerMode set to On
[ INFO] [1646156873.412786664]:  - TriggerSelector set to FrameStart
<font color="#C4A000">[ WARN] [1646156873.412931099]: Feature TriggerSource: value unavailable</font>
	<font color="#C4A000">ERROR Invalid value.</font>
<font color="#C4A000">[ WARN] [1646156873.413378481]:  - Tried to set TriggerSource to FixedRate but the camera used Software instead</font>
[ INFO] [1646156873.414842891]:  - TriggerActivation set to RisingEdge
<font color="#C4A000">[ WARN] [1646156873.414951908]: Could not get feature TriggerDelayAbs, your camera probably doesn&apos;t support it.</font>
<font color="#C4A000">[ WARN] [1646156873.415032872]: Could not get feature ActionDeviceKey, your camera probably doesn&apos;t support it.</font>
<font color="#C4A000">[ WARN] [1646156873.415104273]: Could not get feature ActionGroupKey, your camera probably doesn&apos;t support it.</font>
<font color="#C4A000">[ WARN] [1646156873.415174305]: Could not get feature ActionGroupMask, your camera probably doesn&apos;t support it.</font>
[ INFO] [1646156873.415210964]: Updating Iris config:
<font color="#C4A000">[ WARN] [1646156873.415283593]: Could not get feature IrisAutoTarget, your camera probably doesn&apos;t support it.</font>
<font color="#C4A000">[ WARN] [1646156873.415353305]: Could not get feature IrisMode, your camera probably doesn&apos;t support it.</font>
<font color="#C4A000">[ WARN] [1646156873.415422123]: Could not get feature IrisVideoLevelMax, your camera probably doesn&apos;t support it.</font>
<font color="#C4A000">[ WARN] [1646156873.415493067]: Could not get feature IrisVideoLevelMin, your camera probably doesn&apos;t support it.</font>
<font color="#CC0000">[ERROR] [1646156873.416212084]: Could not start continuous image acquisition. </font>
<font color="#CC0000"> Error: Bad parameter.</font>
[ INFO] [1646156873.416304114]: camera calibration URL: file:///opt/ros/noetic/share/avt_vimba_camera/calibrations/calibration_example.yaml
</pre>

subeiammar, 2022-03-01:
I was having almost the exact same issue (I was using `trigger_source = Line0`) in ROS Melodic.
In my case, `Line0` was not included in the condition to create the frame observer callback [here](https://github.com/astuff/avt_vimba_camera/blob/713a3ad5aa7f5418bc77b8bb75d9cd7c0918a866/src/avt_vimba_camera.cpp#L157). I also found that the [`FrameStartTriggerMode`](https://github.com/astuff/avt_vimba_camera/blob/713a3ad5aa7f5418bc77b8bb75d9cd7c0918a866/include/avt_vimba_camera/avt_vimba_camera.h#L55) enums were not in sync with the [`TriggerMode`](https://github.com/astuff/avt_vimba_camera/blob/713a3ad5aa7f5418bc77b8bb75d9cd7c0918a866/src/avt_vimba_camera.cpp#L44) array.

For @Ronak2904's case, `Software` trigger mode should be implemented, so I don't see why it's causing issues. Maybe the below update mentioned by @panc8170 broke something? I did not have to resolve any compilation issues myself.

> - instead of line_selector I had to rename it to trigger_selector (otherwise the catkin_make would fail)

In any case, I was able to resolve my issue of `Line0` not implemented by updating the `FrameStartTriggerMode` enum values with the `TriggerMode` list, and making the necessary changes [here](https://github.com/astuff/avt_vimba_camera/blob/713a3ad5aa7f5418bc77b8bb75d9cd7c0918a866/src/avt_vimba_camera.cpp#L661) and [here](https://github.com/astuff/avt_vimba_camera/blob/713a3ad5aa7f5418bc77b8bb75d9cd7c0918a866/src/avt_vimba_camera.cpp#L157). Hope this helps someone...

icolwell-as, 2022-03-03:
@subeiammar thanks for posting this! would you be willing to open a PR with the fixes you've found?
We don't have any USB cameras here, which is likely why this bug slipped through.

@Ronak2904 and @panc8170 have the fixes suggested by @subeiammar resolved your issues?

icolwell-as, 2022-04-08:
Closing since the discussion died off. Feel free to re-open or start a new issue if you still experience issues.

closed 2022-04-08 by icolwell-as

Mkarasneh, 2022-06-07:
> Closing since the discussion died off. Feel free to re-open or start a new issue if you still experience problems.

Hello, I am facing the same issue, I tried implementing the edits @subeiammar did in his fork but it is still not showing images in the topics. here is the output I am receiving:

`started roslaunch server http://nazzal-Legion-S7-15ACH6:35131/

SUMMARY
========

PARAMETERS
 * /camera/acquisition_mode: Continuous
 * /camera/acquisition_rate: 60
 * /camera/balance_ratio_abs: 1.0
 * /camera/balance_ratio_selector: Red
 * /camera/binning_x: 1
 * /camera/binning_y: 1
 * /camera/camera_info_url: file:///home/nazz...
 * /camera/decimation_x: 1
 * /camera/decimation_y: 1
 * /camera/exposure: 50000
 * /camera/exposure_auto: Continuous
 * /camera/exposure_auto_alg: FitRange
 * /camera/exposure_auto_max: 16350
 * /camera/exposure_auto_min: 1000
 * /camera/exposure_auto_outliers: 50000
 * /camera/exposure_auto_rate: 50000
 * /camera/exposure_auto_target: 50000
 * /camera/exposure_auto_tol: 5
 * /camera/frame_id: camera
 * /camera/gain: 0
 * /camera/gain_auto: Continuous
 * /camera/gain_auto_adjust_tol: 5
 * /camera/gain_auto_max: 32
 * /camera/gain_auto_min: 0
 * /camera/gain_auto_outliers: 0
 * /camera/gain_auto_rate: 50
 * /camera/gain_auto_target: 50
 * /camera/guid: DEV_1AB22C000E90
 * /camera/height: 1088
 * /camera/ip: 
 * /camera/iris_auto_target: 50
 * /camera/iris_mode: Continuous
 * /camera/iris_video_level_max: 110
 * /camera/iris_video_level_min: 110
 * /camera/line_mode: Input
 * /camera/line_selector: Line0
 * /camera/offset_x: 0
 * /camera/offset_y: 0
 * /camera/pixel_format: RGB8
 * /camera/ptp_mode: Off
 * /camera/show_debug_prints: True
 * /camera/stream_bytes_per_second: 45000000
 * /camera/sync_in_selector: SyncIn1
 * /camera/sync_out_polarity: Normal
 * /camera/sync_out_selector: SyncOut1
 * /camera/sync_out_source: GPO
 * /camera/trigger_activation: RisingEdge
 * /camera/trigger_delay: 0.0
 * /camera/trigger_mode: On
 * /camera/trigger_selector: FrameStart
 * /camera/trigger_source: Software
 * /camera/whitebalance_auto: Continuous
 * /camera/whitebalance_auto_rate: 100
 * /camera/whitebalance_auto_tol: 5
 * /camera/width: 1456
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
[... 70 lines cut ...]
[ WARN] [1654629258.128772206]: Could not get feature PtpMode, your camera probably doesn't support it.
[ INFO] [1654629258.128797062]: Updating PixelFormat config:
[ INFO] [1654629258.284137558]:  - PixelFormat set to Mono8
[ INFO] [1654629258.284194033]: Updating Acquisition and Trigger config:
[ INFO] [1654629258.285721320]:  - AcquisitionMode set to Continuous
[ WARN] [1654629258.285832206]: Feature AcquisitionFrameRate is not writable.
[ WARN] [1654629258.286140010]:  - Tried to set AcquisitionFrameRate to 10 but the camera used 19.9135 instead
[ INFO] [1654629258.287341690]:  - TriggerMode set to On
[ INFO] [1654629258.287654583]:  - TriggerSelector set to FrameStart
[ WARN] [1654629258.287920518]: Feature TriggerSource: value unavailable
	ERROR Invalid value.
[ WARN] [1654629258.288250684]:  - Tried to set TriggerSource to FixedRate but the camera used Software instead
[ INFO] [1654629258.289087723]:  - TriggerActivation set to RisingEdge
[ INFO] [1654629258.290464549]:  - TriggerDelay set to 0
[ WARN] [1654629258.290510505]: Could not get feature ActionDeviceKey, your camera probably doesn't support it.
[ WARN] [1654629258.290537545]: Could not get feature ActionGroupKey, your camera probably doesn't support it.
[ WARN] [1654629258.290560117]: Could not get feature ActionGroupMask, your camera probably doesn't support it.
[ INFO] [1654629258.290580916]: Updating Iris config:
[ WARN] [1654629258.290678508]: Could not get feature IrisAutoTarget, your camera probably doesn't support it.
[ WARN] [1654629258.290731988]: Could not get feature IrisMode, your camera probably doesn't support it.
[ WARN] [1654629258.290769678]: Could not get feature IrisVideoLevelMax, your camera probably doesn't support it.
[ WARN] [1654629258.290798041]: Could not get feature IrisVideoLevelMin, your camera probably doesn't support it.
[ INFO] [1654629258.295466792]: Starting continuous image acquisition ...
[ INFO] [1654629258.295636728]: camera calibration URL: file:///home/nazzal/Project/catkin_ws/src/avt_vimba_camera/calibrations/calibration_example.yaml
`

icolwell-as, 2022-06-08:
Hi @Mkarasneh, were you able to get the camera to work with Vimba Viewer?
That is usually the first step when debugging these camera issues. If it works with vimba viewer, then I suggest running the ROS driver with the same settings that worked in Vimba Viewer.

For example, did you try changing the `trigger_source` to `FixedRate`?

Mkarasneh, 2022-06-18:
> Hi @Mkarasneh, were you able to get the camera to work with Vimba Viewer? That is usually the first step when debugging these camera issues. If it works with vimba viewer, then I suggest running the [...]

Hello @icolwell-as, thank you for your response, the camera worked first with Vimba viewer, then I did what you suggested [here](https://github.com/astuff/avt_vimba_camera/issues/97#issuecomment-1148792219). It seems many of the features of the USB camera are different that the other ones, so I started changing the features in `AvtVimbaCamera.cfg` then changing the corresponding function in `avt_vimba_camera.cpp` then I run `catkin_make` again and test the results.

Actually, with this camera, the `trigger_source` must be `Software`, I also changed `trigger_selector` and `trigger_activation`. Some features were missing, such as `AcquisitionFrameRateEnable` -this makes the `acquisition_rate` writable- and `ExposureMode`, `ExposureActiveMode`, `GainSelector`. Also, the feature named `stream_bytes_per_second` is named `DeviceLinkThroughputLimit` in this camera. Additionally, I removed the features that the camera didn't support, they are shown in my output above.

For anyone testing with this camera, you may need to add more features to your code, simply go the Vimba viewer and see what features you need to change and modify your code accordingly.

here is my working output:

```
NODES
  /
    avt_vimba_nodelet_manager (nodelet/nodelet)
    camera (nodelet/nodelet)
    image_view (image_view/image_view)

auto-starting new master
process[master]: started with pid [40018]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5a9b90e4-ef25-11ec-8d5f-cb9f1aed4610
process[rosout-1]: started with pid [40043]
started core service [/rosout]
process[avt_vimba_nodelet_manager-2]: started with pid [40050]
process[camera-3]: started with pid [40051]
process[image_view-4]: started with pid [40052]
[ INFO] [1655570425.937396302]: Loading nodelet /camera of type avt_vimba_camera/MonoCameraNodelet to manager avt_vimba_nodelet_manager with the following remappings:
[ INFO] [1655570425.940516608]: waitForService: Service [/avt_vimba_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1655570425.944397121]: Initializing nodelet with 16 worker threads.
[ INFO] [1655570425.955993404]: Initializing nodelet with 16 worker threads.
[ INFO] [1655570425.961625361]: waitForService: Service [/avt_vimba_nodelet_manager/load_nodelet] is now available.
[ INFO] [1655570425.995560905]: [Vimba System]: AVT Vimba System initialized successfully
[ INFO] [1655570425.995656726]: Searching for cameras ...
[ INFO] [1655570425.996325163]: Found camera named Allied Vision 1800 U-158c:
[ INFO] [1655570425.996348140]:  - Model Name     : 1800 U-158c
[ INFO] [1655570425.996360921]:  - Camera ID      : DEV_1AB22C000E90
[ INFO] [1655570425.996371537]:  - Serial Number  : 002VK
[ INFO] [1655570425.996382432]:  - Interface ID   : VimbaUSBInterface_0x0
[ INFO] [1655570425.996394793]:  - Interface type : USB
[ INFO] [1655570425.996406387]:  - Access type    : Read and write access
[ INFO] [1655570426.021800794]: Using transport "raw"
[ INFO] [1655570426.064324408]: Trying to open camera by ID: DEV_1AB22C000E90
[ INFO] [1655570426.159072326]: Opened connection to camera named Allied Vision 1800 U-158c with ID DEV_1AB22C000E90
[ INFO] [1655570428.167177018]: Updating Exposure config:
[ WARN] [1655570428.172517464]:  - Tried to set ExposureTime to 5000 but the camera used 5010.63 instead
[ INFO] [1655570428.175293669]:  - ExposureAuto set to Continuous
[ INFO] [1655570428.177745328]:  - ExposureActiveMode set to FlashWindow
[ INFO] [1655570428.179400813]:  - ExposureMode set to Timed
[ INFO] [1655570428.179639246]: Updating Gain config:
[ INFO] [1655570428.181595251]:  - Gain set to 0
[ INFO] [1655570428.183436021]:  - GainAuto set to Continuous
[ INFO] [1655570428.184493886]:  - GainAutoMax set to 24
[ INFO] [1655570428.185809739]:  - GainAutoMin set to 0
[ INFO] [1655570428.186105860]:  - GainSelector set to All
[ INFO] [1655570428.186154119]: Updating White Balance config:
[ INFO] [1655570428.187745979]:  - BalanceRatio set to 1
[ INFO] [1655570428.187942579]:  - BalanceRatioSelector set to Red
[ INFO] [1655570428.189174973]:  - BalanceWhiteAuto set to Continuous
[ INFO] [1655570428.189352087]: Updating Image Mode config:
[ INFO] [1655570428.192473650]:  - BinningHorizontal set to 1
[ INFO] [1655570428.195551424]:  - BinningVertical set to 1
[ INFO] [1655570428.195597518]: Updating ROI config:
[ INFO] [1655570428.199247490]:  - Width set to 1456
[ INFO] [1655570428.202906541]:  - Height set to 1088
[ INFO] [1655570428.207035963]:  - OffsetX set to 0
[ INFO] [1655570428.211162662]:  - OffsetY set to 0
[ INFO] [1655570428.211236622]: Updating Bandwidth config:
[ INFO] [1655570428.213955349]:  - DeviceLinkThroughputLimit set to 450000000
[ INFO] [1655570428.214033500]: Updating GPIO config:
[ INFO] [1655570428.214068280]: Updating USB GPIO config:
[ INFO] [1655570428.214607025]:  - LineSelector set to Line0
[ INFO] [1655570428.215805337]:  - LineMode set to Input
[ INFO] [1655570428.215845076]: Updating PTP config:
[ INFO] [1655570428.215865958]: Updating PixelFormat config:
[ INFO] [1655570428.374407018]:  - PixelFormat set to BayerRG8
[ INFO] [1655570429.026555446]:  - SensorBitDepth set to Adaptive
[ INFO] [1655570429.026668866]: Updating Acquisition and Trigger config:
[... cut ...]
[ INFO] [1655570429.033225015]:  - TriggerSelector set to FrameStart
[ INFO] [1655570429.034455384]:  - TriggerSource set to Software
[ INFO] [1655570429.035529801]:  - TriggerActivation set to RisingEdge
[ INFO] [1655570429.037426722]:  - TriggerDelay set to 0
[ INFO] [1655570429.037485806]: Updating Iris config:
[ INFO] [1655570429.044545570]: Starting continuous image acquisition ...
[ INFO] [1655570429.044738817]: camera calibration URL: file:///home/nazzal/Project/catkin_ws/src/avt_vimba_camera/calibrations/calibration_example.yaml
```

icolwell-as, 2022-06-20:
Hi @Mkarasneh, thanks for sharing your findings!
It sounds similar to [this issue](https://github.com/astuff/avt_vimba_camera/issues/86).
I'm glad you found a working solution! 

I'd suggest opening PRs to better support USB cameras, but I'm not sure if it would be worth the time, especially since the ROS2 driver doesn't have these problems.
Feel free to add a small comment to that other issue I linked with a link to your fork, so that others can find it in case they come looking for better ROS1 USB camera support.

jstumpin, 2022-06-22:
Switching this: [`<arg name="trigger_mode" default="On"/>`](https://github.com/astuff/avt_vimba_camera/blob/ros1_master/launch/mono_camera.launch#L22) to "Off" does it for me.

sarahsallah, 2022-07-04:
Hi,. I'm having an error in launching the usb_avt_vimba camera. After roslauch with guid number, the error is like this:
AVT Vimba System initialized successfully
[ INFO] [1656900604.344615766]: Searching for cameras ...
[ INFO] [1656900604.346697811]: Found camera named Allied Vision 1800 U-507c:
[ INFO] [1656900604.346817113]:  - Model Name     : 1800 U-507c
[ INFO] [1656900604.346908797]:  - Camera ID      : DEV_1AB22C00DAD3
[ INFO] [1656900604.346959231]:  - Serial Number  : 01783
[ INFO] [1656900604.347042563]:  - Interface ID   : VimbaUSBInterface_0x0
[ INFO] [1656900604.347094629]:  - Interface type : USB
[ INFO] [1656900604.347143879]:  - Access type    : Read and write access
[ INFO] [1656900604.649865369]: Trying to open camera by ID: DEV_1AB22C00DAD3
[ INFO] [1656900605.029229547]: Opened connection to camera named Allied Vision 1800 U-507c with ID DEV_1AB22C00DAD3
[ WARN] [1656900607.031899451]: Feature DeviceTemperatureSelector: value unavailable
	ERROR Invalid value.
[ WARN] [1656900607.032493621]: Could not get feature GevTimestampTickFrequency
[ INFO] [1656900607.032732064]: Updating Exposure config:
[ WARN] [1656900607.032988971]: Could not get feature ExposureTimeAbs, your camera probably doesn't support it.
[ INFO] [1656900607.043596613]:  - ExposureAuto set to Continuous
[ WARN] [1656900607.044380808]: Could not get feature ExposureAutoAlg, your camera probably doesn't support it.
[ WARN] [1656900607.044834492]: Could not get feature ExposureAutoAdjustTol, your camera probably doesn't support it.
[ WARN] [1656900607.046743218]: Could not get feature ExposureAutoOutliers, your camera probably doesn't support it.
[ WARN] [1656900607.046998717]: Could not get feature ExposureAutoRate, your camera probably doesn't support it.
[ WARN] [1656900607.047219687]: Could not get feature ExposureAutoTarget, your camera probably doesn't support it.
[ INFO] [1656900607.047418416]: Updating Gain config:
[ INFO] [1656900607.059087641]:  - Gain set to 0
[ INFO] [1656900607.073621986]:  - GainAuto set to Continuous
[ WARN] [1656900607.073923152]: Could not get feature GainAutoAdjustTol, your camera probably doesn't support it.
[ WARN] [1656900607.075241643]:  - Tried to set GainAutoMax to 32 but the camera used 24 instead
[ INFO] [1656900607.078304019]:  - GainAutoMin set to 0
[ WARN] [1656900607.079045844]: Could not get feature GainAutoOutliers, your camera probably doesn't support it.
[ WARN] [1656900607.079399332]: Could not get feature GainAutoRate, your camera probably doesn't support it.
[ WARN] [1656900607.079715698]: Could not get feature GainAutoTarget, your camera probably doesn't support it.
[ INFO] [1656900607.079813943]: Updating White Balance config:
[ WARN] [1656900607.080448979]: Could not get feature BalanceRatioAbs, your camera probably doesn't support it.
[ INFO] [1656900607.081973015]:  - BalanceRatioSelector set to Red
[ INFO] [1656900607.090509621]:  - BalanceWhiteAuto set to Continuous
[ WARN] [1656900607.091276759]: Could not get feature BalanceWhiteAutoAdjustTol, your camera probably doesn't support it.
[ INFO] [1656900607.091909235]: Updating Image Mode config:
[ WARN] [1656900607.092227137]: Could not get feature DecimationHorizontal, your camera probably doesn't support it.
[ WARN] [1656900607.092541583]: Could not get feature DecimationVertical, your camera probably doesn't support it.
[ INFO] [1656900607.101491263]:  - BinningHorizontal set to 1
[ INFO] [1656900607.109805618]:  - BinningVertical set to 1
[ INFO] [1656900607.110106240]: Updating ROI config:
[ INFO] [1656900607.126574592]: Updating Bandwidth config:
[ WARN] [1656900607.127629519]: Could not get feature StreamBytesPerSecond, your camera probably doesn't support it.
[ INFO] [1656900607.127789238]: Updating GPIO config:
[ WARN] [1656900607.128023840]: Could not get feature SyncInSelector, your camera probably doesn't support it.
[ WARN] [1656900607.128209481]: Could not get feature SyncOutPolarity, your camera probably doesn't support it.
[ WARN] [1656900607.128394385]: Could not get feature SyncOutSelector, your camera probably doesn't support it.
[ WARN] [1656900607.129682602]: Could not get feature SyncOutSource, your camera probably doesn't support it.
[ INFO] [1656900607.130133886]: Updating USB GPIO config:
[ WARN] [1656900607.131325204]: Feature LineSelector: value unavailable
	ERROR Invalid value.
[ WARN] [1656900607.131934063]:  - Tried to set LineSelector to FrameStart but the camera used Line0 instead
[ INFO] [1656900607.138283627]:  - LineMode set to Input
[ INFO] [1656900607.138660859]: Updating PTP config:
[ WARN] [1656900607.138829475]: Could not get feature PtpMode, your camera probably doesn't support it.
[ INFO] [1656900607.138963017]: Updating PixelFormat config:
[ INFO] [1656900607.143829058]: Updating Acquisition and Trigger config:
[ WARN] [1656900607.145682005]: Could not get feature AcquisitionFrameRateAbs, your camera probably doesn't support it.
[ WARN] [1656900607.147960603]: Feature TriggerSelector: value unavailable
	ERROR Invalid value.
[ WARN] [1656900607.148410831]:  - Tried to set TriggerSelector to Line0 but the camera used FrameStart instead
[ WARN] [1656900607.149337976]: Feature TriggerSource: value unavailable
	ERROR Invalid value.
[ WARN] [1656900607.152818676]:  - Tried to set TriggerSource to FixedRate but the camera used Software instead
[ INFO] [1656900607.157530246]:  - TriggerActivation set to RisingEdge
[... cut ...]
[ WARN] [1656900607.160898013]: Could not get feature IrisMode, your camera probably doesn't support it.
[ WARN] [1656900607.161422260]: Could not get feature IrisVideoLevelMax, your camera probably doesn't support it.
[ WARN] [1656900607.161584443]: Could not get feature IrisVideoLevelMin, your camera probably doesn't support it.
[ INFO] [1656900607.181588857]: Starting continuous image acquisition ...
[ INFO] [1656900607.182124497]: camera calibration URL: file:///home/agx0/catkin_ws/src/avt_vimba_camera/calibrations/calibration_example.yaml

Please help.. Thank you!
:

Mkarasneh, 2022-07-04:
> Hi @Mkarasneh, thanks for sharing your findings! It sounds similar to [this issue](https://github.com/astuff/avt_vimba_camera/issues/86). I'm glad you found a working solution!
>
> I'd suggest opening [...]

Thank you @icolwell-as, I don't think my code is tidy enough to make a PR, but I'll add a comment to the other issue and a link to my fork in case someone wants to use or edit it.

Mkarasneh, 2022-07-04:
> Hi,. I'm having an error in launching the usb_avt_vimba camera. After roslauch with guid number, the error is like this: AVT Vimba System initialized successfully [ INFO] [1656900604.344615766]: [...]

This is the same as the issue I was facing, see my comment above on how I fixed it. I will create a fork with my code tomorrow so that you and everybody can use it. I hope it'll solve the errors you are facing now.

sarahsallah, 2022-07-04:
> This is the same as the issue I was facing, see my comment above on how I fixed it. I will create a fork with my code tomorrow so that you and everybody can use it. I hope it'll solve the errors you [...]

Please help me to solve this issue.  i can be reach at: sitisarah.mdsallah@gmail.com

Mkarasneh, 2022-07-04:
> > > Hi,. I'm having an error in launching the usb_avt_vimba camera. After roslauch with guid number, the error is like this: AVT Vimba System initialized successfully [ INFO] [1656900604.344615766]: Searching for cameras ... [ INFO] [1656900604.346697811]: Found camera named Allied Vision 1800 U-507c: [ INFO] [1656900604.346817113]: - Model Name : 1800 U-507c [ INFO] [1656900604.346908797]: - Camera ID : DEV_1AB22C00DAD3 [ INFO] [1656900604.346959231]: - Serial Number : 01783 [ INFO] [1656900604.347042563]: - Interface ID : VimbaUSBInterface_0x0 [ INFO] [1656900604.347094629]: - Interface type : USB [ INFO] [1656900604.347143879]: - Access type : Read and write access [ INFO] [1656900604.649865369]: Trying to open camera by ID: DEV_1AB22C00DAD3 [ INFO] [1656900605.029229547]: Opened connection to camera named Allied Vision 1800 U-507c with ID DEV_1AB22C00DAD3 [ WARN] [1656900607.031899451]: Feature DeviceTemperatureSelector: value unavailable ERROR Invalid value. [ WARN] [1656900607.032493621]: Could not get feature GevTimestampTickFrequency [ INFO] [1656900607.032732064]: Updating Exposure config: [ WARN] [1656900607.032988971]: Could not get feature ExposureTimeAbs, your camera probably doesn't support it. [ INFO] [1656900607.043596613]: - ExposureAuto set to Continuous [ WARN] [1656900607.044380808]: Could not get feature ExposureAutoAlg, your camera probably doesn't support it. [ WARN] [1656900607.044834492]: Could not get feature ExposureAutoAdjustTol, your camera probably doesn't support it. [ WARN] [1656900607.046743218]: Could not get feature ExposureAutoOutliers, your camera probably doesn't support it. [ WARN] [1656900607.046998717]: Could not get feature ExposureAutoRate, your camera probably doesn't support it. [ WARN] [1656900607.047219687]: Could not get feature ExposureAutoTarget, your camera probably doesn't support it. [ INFO] [1656900607.047418416]: Updating Gain config: [ INFO] [1656900607.059087641]: - Gain set to 0 [ INFO] [1656900607.073621986]: - GainAuto set to Continuous [ WARN] [1656900607.073923152]: Could not get feature GainAutoAdjustTol, your camera probably doesn't support it. [ WARN] [1656900607.075241643]: - Tried to set GainAutoMax to 32 but the camera used 24 instead [ INFO] [1656900607.078304019]: - GainAutoMin set to 0 [ WARN] [1656900607.079045844]: Could not get feature GainAutoOutliers, your camera probably doesn't support it. [ WARN] [1656900607.079399332]: Could not get feature GainAutoRate, your camera probably doesn't support it. [ WARN] [1656900607.079715698]: Could not get feature GainAutoTarget, your camera probably doesn't support it. [ INFO] [1656900607.079813943]: Updating White Balance config: [ WARN] [1656900607.080448979]: Could not get feature BalanceRatioAbs, your camera probably doesn't support it. [ INFO] [1656900607.081973015]: - BalanceRatioSelector set to Red [ INFO] [1656900607.090509621]: - BalanceWhiteAuto set to Continuous [ WARN] [1656900607.091276759]: Could not get feature BalanceWhiteAutoAdjustTol, your camera probably doesn't support it. [ INFO] [1656900607.091909235]: Updating Image Mode config: [ WARN] [1656900607.092227137]: Could not get feature DecimationHorizontal, your camera probably doesn't support it. [ WARN] [1656900607.092541583]: Could not get feature DecimationVertical, your camera probably doesn't support it. [ INFO] [1656900607.101491263]: - BinningHorizontal set to 1 [ INFO] [1656900607.109805618]: - BinningVertical set to 1 [ INFO] [1656900607.110106240]: Updating ROI config: [ INFO] [1656900607.126574592]: Updating Bandwidth config: [ WARN] [1656900607.127629519]: Could not get feature StreamBytesPerSecond, your camera probably doesn't support it. [ INFO] [1656900607.127789238]: Updating GPIO config: [ WARN] [1656900607.128023840]: Could not get feature SyncInSelector, your camera probably doesn't support it. [ WARN] [1656900607.128209481]: Could not get feature SyncOutPolarity, your camera probably doesn't support it. [ WARN] [1656900607.128394385]: Could not get feature SyncOutSelector, your camera probably doesn't support it. [ WARN] [1656900607.129682602]: Could not get feature SyncOutSource, your camera probably doesn't support it. [ INFO] [1656900607.130133886]: Updating USB GPIO config: [ WARN] [1656900607.131325204]: Feature LineSelector: value unavailable ERROR Invalid value. [ WARN] [1656900607.131934063]: - Tried to set LineSelector to FrameStart but the camera used Line0 instead [ INFO] [1656900607.138283627]: - LineMode set to Input [ INFO] [1656900607.138660859]: Updating PTP config: [ WARN] [1656900607.138829475]: Could not get feature PtpMode, your camera probably doesn't support it. [ INFO] [1656900607.138963017]: Updating PixelFormat config: [ INFO] [1656900607.143829058]: Updating Acquisition and Trigger config: [ WARN] [1656900607.145682005]: Could not get feature AcquisitionFrameRateAbs, your camera probably doesn't support it. [ WARN] [1656900607.147960603]: Feature TriggerSelector: value unavailable ERROR Invalid value. [ WARN] [1656900607.148410831]: - Tried to set TriggerSelector to Line0 but the camera used FrameStart instead [ WARN] [1656900607.149337976]: Feature TriggerSource: value unavailable ERROR Invalid value. [ WARN] [1656900607.152818676]: - Tried to set TriggerSource to FixedRate but the camera used Software instead [ INFO] [1656900607.157530246]: - TriggerActivation set to RisingEdge [ WARN] [1656900607.157944313]: Could not get feature TriggerDelayAbs, your camera probab [... cut ...]
> > This is the same as the issue I was facing, see my comment above on how I fixed it. I will create a fork with my code tomorrow so that you and everybody can use it. I hope it'll solve the errors you are facing now.
> 
> Please help me to solve this issue. i can be reach at: [sitisarah.mdsallah@gmail.com](mailto:sitisarah.mdsallah@gmail.com)

Hello @sarahsallah, you can view my comment on the other [issue](https://github.com/astuff/avt_vimba_camera/issues/86) for details on how to fix your issue. Thank you!