astuff/avt_vimba_camera #80: G-319 Frame rate issues

JonasLos, 2022-03-01:
I'm running two G-319 with this driver - the Frame rate is set to 35 Hz, the driver says it can "only set it to 33" which is fine. After just a few seconds the frame rate starts to drop until it eventually gets to about 3.6 Hz (according to rostopic hz). 

Is this a known issue?

Thank you

icolwell-as, 2022-03-03:
Hi @JonasLos, the frame rate dropping is actually very typical, and the usual cause is exposure settings. I recommend playing with the camera settings using VimbaViewer to find settings that result in the framerate you desire, then save these settings into a launch file or params yaml to use with the ROS driver.

JonasLos, 2022-03-03:
Thank you, makes perfect sense. I was looking at that in the viewer and could see it was jumping around a lot even in a room with stable light conditions so that certainly makes sense. 

Appreciate the quick response.

closed 2022-03-03 by JonasLos

JonasLos, 2022-05-04:
Hello, 

after some back and forth and again getting very low FPS out of the cameras (even after setting the exposure and gain values manually) I found the actual issue that was causing this: we had an out of the box rviz setup that was subscribing to image_raw/theora. Simply changing it to raw or compressed completely alleviated the low FPS and we were getting the targeted 30 FPS with Auto Exposure/Gain on.

icolwell-as, 2022-05-04:
Woah, that's bizarre, never seen that before. So `rostopic hz /image_raw` would give you low FPS when RViz was running? Even though RViz is subscribed to `/image_raw/theora`?
Thanks for sharing your findings!

JonasLos, 2022-05-04:
Correct, we have 3 other cameras (lucid) and 2 makos and the launch file starting all sensors was opening rviz with all image topics subscribed to /image_raw/theora for the respective cameras. I suppose theora might be processed on the computer itself as that seems to be at a lower FPS than raw or compressed in general? 
Also, the same happens when recording a rosbag without increasing the buffer size. For example running rosbag record -a instead of rosbag record -a ----buffsize=0 (0 = infinite).
 I suppose a seperate question: 
Is there a way of turning off the theora publishing from this node?

icolwell-as, 2022-05-04:
Interesting. I looked into it a bit more, see info here: http://wiki.ros.org/theora_image_transport
I'm guessing that as soon as a subscriber appears for the /theora topic, then the theora_image_transport plugin will start encoding into theora format, then RViz immediately decodes it to display it.

You may be able to disable the theora topic completely by removing the plugin from your system:
`sudo apt remove ros-$ROS_DISTRO-theora-image-transport`