astuff/avt_vimba_camera #82: Problem with camera info message

gustavojoseleite, 2022-03-09:
I'm using a mako G319 camera whose maximum resolution is 1536x2064. I applied binning to reduce the resolution by four times, that is, to 384x516. When performing the calibration of the set (camera and lenses) the calibration parameters were loaded and the camera info of the camera driver is published as follows:
height: 1544
width: 2064
distortion_model: "plumb_bob"
D: [-0.097715, 0.08086, -0.003701, 0.001999, 0.0]
K: [299.54396, 0.0, 254.14465, 0.0, 300.14093, 183.94307, 0.0, 0.0, 1.0]
A: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [298.52789, 0.0, 255.89954, 0.0, 0.0, 300.0076, 182.47257, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 4
binning_y: 4

As a result, the rectified image is distorted and approximately one-quarter of the actual size. I created a new publish for the camera info with the resolution considering the binning applied as follows:
height: 384
width: 516
distortion_model: "plumb_bob"
D: [-0.097715, 0.08086, -0.003701, 0.001999, 0.0]
K: [299.54396, 0.0, 254.14465, 0.0, 300.14093, 183.94307, 0.0, 0.0, 1.0]
A: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [298.52789, 0.0, 255.89954, 0.0, 0.0, 300.0076, 182.47257, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0

After this last change, the image was correctly rectified and was as expected. Is there a way to publish the correct camera info without having to republish it by changing resolution and binning? Could there be an error during the calibration process that led to this issue?

icolwell-as, 2022-03-09:
Hi @gustavojoseleite, according to REP-104, the first camera info example you posted is correct. Please see the following documentation that explains how the camera info topic should be populated:

https://www.ros.org/reps/rep-0104.html#camerainfo-message
http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html

Maybe you need to fix whatever is subscribing to the camera info, to take into account the binning fields in the message.

gustavojoseleite, 2022-03-16:
Thanks for your reply @icolwell-as. Image proc is subscribing this camera info and publishing a wrong image for the first case, in the second case the image published is right. I will check your links.

closed 2022-03-16 by gustavojoseleite