astuff/avt_vimba_camera #85: Can find the GUID of the camera.

samdestroy3, 2022-03-10:
Hi,

I'm pretty new to ros.
I'm currently working with the following camera: alvium U-319C.
When I try to roslaunch  mono_camera.launch, the following error is showing: Can't connect to the camera: at least GUID or IP need to be set.

Following one or two tutorial, i found that the GUID could be the serial number , or that the GUID can be found with vimba Viewer. 
The issue with that the S/N is only 5 character long and does not solve the problem. 
The other issues is that the Vimba Viewer does not show the GUID.

Some information about how I can find the GUID would be appreciated.
Link to tutorial that could help me to understand more how to work with this package would also be nice.

Thank you for your time and your support,
Sam

icolwell-as, 2022-03-11:
Hi @samdestroy3, I don't use USB cameras (only ethernet cameras), but when I open Vimba Viewer, I can see the GUID here:
[image]

In my case, the GUID is `DEV_000F315C353E`.

samdestroy3, 2022-03-13:
Thank you!
It is working.