astuff/avt_vimba_camera #86: ROS1: USB Camera Feature Names

subeiammar, 2022-03-11:
I'm using an Alvium 1800 U-1236c USB camera in ROS Melodic. I noticed that some feature configuration names are incorrect in the main [`avt_vimba_camera.cpp`](https://github.com/astuff/avt_vimba_camera/blob/ros1_master/src/avt_vimba_camera.cpp) source file.

When the node tries to configure the camera, ROS warning messages pop up for certain features like:
```
Could not get feature AcquisitionFrameRateAbs, your camera probably doesn't support it.
Could not get feature TriggerDelayAbs, your camera probably doesn't support it.
Could not get feature ExposureTimeAbs, your camera probably doesn't support it.
```
I'm not sure if feature names differ between the different cameras, but [Alvium's documentation](https://www.alliedvision.com/en/support/technical-documentation/alvium-usb-documentation/) refer to some feature names that conflict with those in the [config functions in `avt_vimba_camera.cpp`](https://github.com/astuff/avt_vimba_camera/blob/713a3ad5aa7f5418bc77b8bb75d9cd7c0918a866/src/avt_vimba_camera.cpp#L863).
e.g `AcquisitionFrameRateAbs -> AcquisitionFrameRate` and `ExposureTimeAbs -> ExposureTime`

subeiammar, 2022-03-16:
Thanks for looking into it!
My Alvium U-1236c camera had firmware version `6.0.35992`, and you can check 4024d68cc275c88b5c6e9cd0a4f81f8bdb8bc7fd out for the features that I found were different. Note that there's even a difference in the `DeviceTemperatureSelector` feature **_value_**.

Updating my firmware version to `7.0.81DB3896` didn't help with the feature names, however it did fix a separate issue I was having with updating camera frame rate so that's a plus! :upside_down_face:

Unfortunately we're dependent on other sensor drivers that use ROS1, but I will definitely look further into making the switch later. Thanks again for your responses and help!

icolwell-as, 2022-03-17:
No problem! glad the firmware update fixed something at least :smile: 

Ya, sounds like your best option is to use your own modified driver as you are doing. 
If enough people use the Alvium series cameras, I would be open to adding some official support for the varying feature names.
Or if someone opens a PR to add alvium support, I'll review and merge as-needed. I imagine some kind of feature name hashtable/map could be used to hold these differences in feature names.

Glad you have a solution for now! I'll leave this issue open in case any other Alvium users come here with similar issues.

jstumpin, 2022-06-22:
Hear, hear! Alvium 1800 U-319c user here. Firmware: 00.08.00.6727174b. Vimba SDK: v6.0. Same issue.

Mkarasneh, 2022-07-04:
Hello,

I am using the Alvium 1800 U-158c, Firmware: 00.08.00.6727174b. Vimba SDK: v6.0. and I used the fork from @subeiammar as a starting point and edited the features names as I pointed out in [this issue](https://github.com/astuff/avt_vimba_camera/issues/79).

I don't think the code is organized enough to create a PR. However, it is working so that anyone how's trying to get the camera to work can use it, see my fork [here](https://github.com/Mkarasneh/avt_vimba_camera). You only need to update the configurations in the `mono_camera.launch` file, if -for any reason- those configurations were not applied, try changing the configuration in the `cfg/AvtVimbaCamera.cfg` file.