astuff/avt_vimba_camera #88: Problems using camera in diferrent light conditions

gustavojoseleite, 2022-03-17:
Hi 
I'm working with day and night camera settings. I kept the limits for exposure between 0-100000 (as the documentation indicates, I need to keep a constant frame rate of 10hz) and for the gain between 0-30 (although it is not recommended to keep a high value for the gain, as it increases the noise ).
For the exposure algorithm I was using FitRange, but I was unable to obtain, given the frame rate restrictions, good parameters for night use (image was too dark) I ended up changing to the Mean algorithm which, despite presenting high gains for the night environment, provides clearer images.

In some cases, during the light flicker, the camera image was completely white and even after some time, it didn't return to a good setting (I'm keeping the continuous adjustment for exposure, gain and white balance), it was necessary to restart the driver from the camera. From what I read in the documentation Mean's algorithm can saturate the image in some cases, is there any way to mitigate this problem? Or some other parameter that can be changed in order to get good images for use during the day and night keeping the frame rate constant and avoiding the noise problem?

icolwell-as, 2022-03-17:
Hi @gustavojoseleite, it sounds like these issues are inherent to the camera itself and not necessarily related to the ROS driver.
Do you see similar behaviour running the camera with VimbaViewer? If you do, then I recommend that you reach out to [AVT support](https://www.alliedvision.com/en/support/).

icolwell-as, 2022-08-10:
Closing due to lack of response.

closed 2022-08-10 by icolwell-as