astuff/avt_vimba_camera #89: Setting params of gamma and hue using launch

gustavojoseleite, 2022-03-17:
Hi,
I need to set the params of gamma and hue for my camera. I could change them using the VimbaViewer but i have acess to the camera just using ssh to the computer connected to her. It is possible to change them using roslaunch? Or using command line?

Thanks.

icolwell-as, 2022-03-17:
Hi @gustavojoseleite, I assume you are using the ROS1 driver, is that true?
The ROS1 driver only exposes camera features listed in the cfg file here:
https://github.com/astuff/avt_vimba_camera/blob/ros1_master/cfg/AvtVimbaCamera.cfg

The ROS2 driver is different in that it queries the camera and generates ROS parameters for every single camera feature available (everything you see in VimbaViewer).

If you want to add hue and gamma parameters to the ROS1 driver, feel free to open a PR and I can review.

icolwell-as, 2022-08-10:
Gamma is being added by PR #101