astuff/avt_vimba_camera #94: Nodelet Launch file does not properly configure camera

bonjiman, 2022-04-20:
Hi, 

I'm working with the Vimba ROS1 driver on a "real-time" image processing system.  I have a lot of processing tasks to do, so I'd like to be able to utilize the Nodelet version of the Vimba driver in order to minimize the slowdown from de/serialization.  I've successfully been able to edit and launch the non-nodelet launch files, but the nodelet versions do not seem to work for me.  Whenever I run the `mono_camera_nodelet.launch` file, the camera launches using the default settings from the `cfg/AvtVimbaCamera.cfg` file instead of using the settings provided in the launch file.  This issue does not occur using the non-nodelet .launch files.

Any idea what I could be doing differently?

Thanks

icolwell-as, 2022-04-20:
Hi @bonjiman, what version of the driver are you using? and what version of ROS are you running?

Thanks for pointing out the issue. I tried launching the nodelet launch file and you're right, there does seem to be an issue.
If I run `rosparam list` after launching, I see two duplicated sets of parameters. One set starts with `/camera` and the other starts with `/avt_vimba_nodelet_manager`.

I edited the nodelet launch file as follows:

```
  <!-- The mono camera nodelet -->
  <node pkg="nodelet" type="nodelet" name="$(arg name)" args="load avt_vimba_camera/MonoCameraNodelet avt_vimba_nodelet_manager" output="screen">
    <param name="ip" value="$(arg ip)"/>
    <param name="guid" value="$(arg guid)"/>
    <param name="frame_id" value="$(arg frame_id)"/>
    <param name="camera_info_url" value="$(arg camera_info_url)"/>
    <param name="show_debug_prints" value="$(arg show_debug_prints)"/>
  </node>
  <group ns="avt_vimba_nodelet_manager">
    <param name="acquisition_mode" value="$(arg acquisition_mode)"/>
    <param name="acquisition_rate" value="$(arg acquisition_rate)"/>

    <param name="trigger_source" value="$(arg trigger_source)"/>
    <param name="trigger_mode" value="$(arg trigger_mode)"/>
    <param name="trigger_selector" value="$(arg trigger_selector)"/>
    <param name="trigger_activation" value="$(arg trigger_activation)"/>
    <param name="trigger_delay" value="$(arg trigger_delay)"/>

    <param name="exposure" value="$(arg exposure)"/>
    <param name="exposure_auto" value="$(arg exposure_auto)"/>
    <param name="exposure_auto_alg" value="$(arg exposure_auto_alg)"/>
    <param name="exposure_auto_tol" value="$(arg exposure_auto_tol)"/>
    <param name="exposure_auto_max" value="$(arg exposure_auto_max)"/>
    <param name="exposure_auto_min" value="$(arg exposure_auto_min)"/>
    <param name="exposure_auto_outliers" value="$(arg exposure_auto_outliers)"/>
    <param name="exposure_auto_rate" value="$(arg exposure_auto_rate)"/>
    <param name="exposure_auto_target" value="$(arg exposure_auto_target)"/>

    <param name="gain" value="$(arg gain)"/>
    <param name="gain_auto" value="$(arg gain_auto)"/>
    <param name="gain_auto_adjust_tol" value="$(arg gain_auto_adjust_tol)"/>
    <param name="gain_auto_max" value="$(arg gain_auto_max)"/>
    <param name="gain_auto_min" value="$(arg gain_auto_min)"/>
    <param name="gain_auto_outliers" value="$(arg gain_auto_outliers)"/>
    <param name="gain_auto_rate" value="$(arg gain_auto_rate)"/>
    <param name="gain_auto_target" value="$(arg gain_auto_target)"/>

    <param name="balance_ratio_abs" value="$(arg balance_ratio_abs)"/>
    <param name="balance_ratio_selector" value="$(arg balance_ratio_selector)"/>
    <param name="whitebalance_auto" value="$(arg whitebalance_auto)"/>
    <param name="whitebalance_auto_tol" value="$(arg whitebalance_auto_tol)"/>
    <param name="whitebalance_auto_rate" value="$(arg whitebalance_auto_rate)"/>

    <param name="binning_x" value="$(arg binning_x)"/>
    <param name="binning_y" value="$(arg binning_y)"/>
    <param name="decimation_x" value="$(arg decimation_x)"/>
    <param name="decimation_y" value="$(arg decimation_y)"/>

    <param name="width" value="$(arg width)"/>
    <param name="height" value="$(arg height)"/>
    <param name="offset_x" value="$(arg offset_x)"/>
    <param name="offset_y" value="$(arg offset_y)"/>

    <param name="pixel_format" value="$(arg pixel_format)"/>

    <param name="stream_bytes_per_second" value="$(arg stream_bytes_per_second)"/>

    <param name="ptp_mode" value="$(arg ptp_mode)"/>

    <param name="sync_in_selector" value="$(arg sync_in_selector)"/>
    <param name="sync_out_polarity" value="$(arg sync_out_polarity)"/>
    <param name="sync_out_selector" value="$(arg sync_out_selector)"/>
    <param name="sync_out_source" value="$(arg sync_out_source)"/>

    <param name="line_selector" value="$(arg line_selector)"/>
    <param name="line_mode" value="$(arg line_mode)"/>

    <param name="iris_auto_target" value="$(arg iris_auto_target)"/>
    <param name="iris_mode" value="$(arg iris_mode)"/>
    <param name="iris_video_level_min" value="$(arg iris_video_level_min)"/>
    <param name="iris_video_level_max" value="$(arg iris_video_level_max)"/>
  </group>
```

Which seems to partially work as a temporary hack, but it's not ideal since the parameters are split across 2 namespaces now. If we can figure out a way to get the dynamic_reconfigure parameters to use the /camera namespace, then I think it would resolve everything.

bonjiman, 2022-04-20:
I just tried it and it worked!!  Thanks so much.  I'm kicking myself for not using `rostopic list` to see that earlier. 

I agree this isn't *ideal* since the parameters would be more appropriate in the /camera namespace, but for now I'm happy to be able to report to my coworkers that it's working.  I'll be able to look at this more over the weekend.  

Thanks again!!  You rock for getting back to me so quickly.

icolwell-as, 2022-08-10:
Resolved by #102

closed 2022-08-10 by icolwell-as