astuff/avt_vimba_camera #100: Setting ROI parameters does not work when the system is started first time

kanwaljahan, 2022-07-21:
Hi,
We are using Mako G-223C for recording videos from running vehicles. The data needs to be recorded automatically from only a defined region of interest whenever the vehicle starts moving. The OS used is Linux 16.04 and the ROS-kinetic branch of avt_vimba_camera.  The problem is ROI settings are not reflected in the recorded videos on the first power of the system. 

- Roi_settings can not be updated once the camera has already started recording, the same behavior is seen in the Vimba_viewer app too.
-  Restarting the system (when the camera still has the power) does reflect the ROI settings.
- When the system is restarted and does reflect the ROI settings, unplugging the camera from the power and replugging it again ends up in the lost ROI settings. 
Any idea what is happening here? I see in https://github.com/astuff/avt_vimba_camera/blob/kinetic/cfg/AvtVimbaCamera.cfg that the sensor level for ROI settings is set to RECONFIGURE_RUNNING. Should it be RECONFIGURE_STOP or RECONFIGURE_CLOSE? I don't know the difference between latter two. Thanks!

icolwell-as, 2022-07-21:
Hi @kanwaljahan, the `RECONFIGURE_*` params are a ROS dynamic reconfigure thing. Based on the ros1_master branch, they aren't actually used. According to the [updateConfig](https://github.com/astuff/avt_vimba_camera/blob/ros1_master/src/avt_vimba_camera.cpp#L788) function, the camera is stopped every time any dynamic reconfigure parameter is changed.
I'm a bit confused trying to understand your use case, so I've got some additional questions:

- Roi_settings can not be updated once the camera has already started recording, the same behavior is seen in the Vimba_viewer app too.
**This behaviour is expected, the video streaming would have to stop and start again to reflect ROI changes. If vimba viewer shows this behaviour, it is likely a limitation of the camera, you may need to reach out to AVT support to get more info/help**
- Restarting the system (when the camera still has the power) does reflect the ROI settings.
**What do you mean by "system"? restarting the launch file for the camera driver? Regardless, this is expected behaviour**
- When the system is restarted and does reflect the ROI settings, unplugging the camera from the power and replugging it again ends up in the lost ROI settings.
**This is also expected behaviour. The camera reboots, so it reverts to default settings. You can save settings using the `UserSet*` features, perhaps that is helpful for saving settings across boots**

The only issue I see here is that the camera doesn't stop and re-start image acquisition when the dynamic reconfigure parameters are changed during streaming. Is that the case? This actually sounds similar to this [other issue](https://github.com/astuff/avt_vimba_camera/issues/97) that mentions problems with stopping/starting image acquisition when parameters are changed.

kanwaljahan, 2022-07-22:
Hi, @icolwell-as thank you for your response. My faced problem is ROI settings are not reflected in the first launch ever, settings are only reflected after relaunch while the camera is still on power. And according to my understanding problem arises because in mono_camera.cpp https://github.com/astuff/avt_vimba_camera topic starts publishing /blob/31f04d41272ab875424b5e31f9374ffec7e88129/src/mono_camera.cpp#L49 and after configurations are updated https://github.com/astuff/avt_vimba_camera/blob/31f04d41272ab875424b5e31f9374ffec7e88129/src/mono_camera.cpp#L65. 
I have seen similar behavior in vimbaviewr app also. In the app, some settings like exposure, and brightness can be adjusted while the camera is already recording but not ROI settings. I need to stop recording, adjust the ROI offsets, and then start the recording again. In my understanding, the same behavior is reflected in the launch process too, since the camera feed is already captured/ topics start to publish the ROI settings are not updated. 
I will explain a little about the use case just for background info, we have automated the collection of image data such that whenever a vehicle is started, installed PC + cameras get the power, and a bash script is called to start the launch file. 
So my short question is: What is the expected behavior about ROI settings on the first launch of mono_camera.launch?

kanwaljahan, 2022-07-22:
I have also tried hardcoding the roi offests in the .cfg file. But that does not help either because of topic publishing already before the call of dynamic reconfigure callback. And roi offsets can not be set once the camera starts recording

icolwell-as, 2022-08-03:
Hi @kanwaljahan, lines 49 and 65 merely set the callbacks, the camera stream doesn't start until line 65 gets called, which then calls line 116 to handle the reconfigure.
To your point though, you are right in that the camera starts streaming first, then gets reconfigured. 
I think there is a bug somewhere though, because it should immediately stop the imaging, re-configure, then re-start the imaging with the correct settings.
I wonder if the `streaming_` variable isn't set to true at first start, which then causes it not to stop imaging on the re-configure.