astuff/avt_vimba_camera #103: configureFeature function

JanezCim, 2022-08-09:
I have a hard time understanding the configureFeature fucntion:

```
void AvtVimbaCamera::configureFeature(const std::string& feature_str, const Vimba_Type& val_in, Std_Type& val_out)
{
  Vimba_Type actual_value;

  VmbErrorType return_value = setFeatureValue(feature_str, val_in);
  if (return_value == VmbErrorSuccess || return_value == VmbErrorInvalidValue)
  {
    getFeatureValue(feature_str, actual_value);
    if (val_in == actual_value)
    {
      ROS_INFO_STREAM(" - " << feature_str << " set to " << actual_value);
    }
    else
    {
      ROS_WARN_STREAM(" - Tried to set " << feature_str << " to " << val_in << " but the camera used " << actual_value
                                         << " instead");
      val_out = static_cast<Std_Type>(actual_value);
    }
  }
  else
  {
    val_out = static_cast<Std_Type>(val_in);
  }
}
```
It does not make sense to me that `val_out` gets set to `val_in` if either of these cases are false `return_value == VmbErrorSuccess || return_value == VmbErrorInvalidValue`. As i understand it, if setting feature on camera was successful, `val_out` must be set to `val_in` BUT if not successful `val_out` must be set to `actual_value`. This does not seem to be the case with this code. 

I've slightly changed this function like:
```
void AvtVimbaCamera::configureFeature(const std::string& feature_str, const Vimba_Type& val_in, Std_Type& val_out)
{
  Vimba_Type actual_value;

  VmbErrorType return_value = setFeatureValue(feature_str, val_in);
  if (return_value != VmbErrorSuccess)
  {
    ROS_WARN_STREAM("Configuring feature " << feature_str << " failed because: "
                                           << api_.errorCodeToMessage(return_value));
    return;
  }

  // double check that the feature was set by asking the camera about actual value
  getFeatureValue(feature_str, actual_value);
  if (val_in == actual_value)
  {
    ROS_INFO_STREAM(" - " << feature_str << " set to " << actual_value);
  // sometimes actual value is not exactly set (eg. ExposureTime instead of 18000 is set to 18002). That is ok.
  }
  else
  {
    ROS_WARN_STREAM(" - Tried to set " << feature_str << " to " << val_in << " but the camera used " << actual_value
                                      << " instead");
    val_out = static_cast<Std_Type>(actual_value);
  }
}
```
Would this be a valid fix suggestion or have i misunderstood the code?

icolwell-as, 2022-10-06:
Hi @JanezCim, I found some time to review the code. I think you have misunderstood the if statement.

> It does not make sense to me that val_out gets set to val_in if either of these cases are false return_value == VmbErrorSuccess || return_value == VmbErrorInvalidValue.

That is not true, the else statement only executes if _both_ of those cases are false, not if _either_ of the case are false. Keep in mind it is an OR (`||`), not an AND (`&&`).

> As i understand it, if setting feature on camera was successful, val_out must be set to val_in BUT if not successful val_out must be set to actual_value.

Nope, if the feature was set successfully, then the `if (return_value == VmbErrorSuccess || return_value == VmbErrorInvalidValue)` statement evaluates true and val_out is set to actual_value.

I hope this makes sense. 
If you have camera hardware, you can also see the code working as expected by running the driver, all the output is displayed in the terminal.

JanezCim, 2022-10-11:
Okay, I see what is the logic now, thank you!

closed 2022-10-11 by JanezCim