astuff/avt_vimba_camera #109: Multiple USB Cameras Possible?

mattberkem, 2022-12-17:
Using ros2_master, USB Allied Vision 1800 U-1240c, Vimba 6.0. Each camera works separately. But when I try to run both simultaneously, only the camera started first will output. This behavior is also seen with Vimba_Viewer, but I am hoping someone here can help. I would like to run 2 cameras simultaneously on the same computer.

icolwell-as, 2023-01-09:
Hi @mattberkem, if the issue exists with the Vimba Viewer desktop app provided by AVT, then the issue is likely outside of the scope of the ROS driver. Your best option is to reach out to AVT support directly:
https://www.alliedvision.com/en/about-us/contact-us/technical-support-repair-/-rma/ 

Good luck!

agruetPL, 2023-02-27:
Hello @mattberkem, I ran into the same problem than you. 
I can't make more than two cameras works at the same time on the same computer. 
My process is to start multiples nodes, one for each of my cameras, but only the first two nodes actually work, nothing is published by the other ones. 
Same thing with Vimba Viewer, only the two cameras I launch first are working, I do not get any images from the other ones.
Did you find a solution to this problem?
Thank you!

dagata-mining, 2023-03-03:
We came accros this problem read this note for more information: https://cdn.alliedvision.com/fileadmin/content/documents/products/cameras/various/appnote/USB/Setting_Up_USB_Vision_Systems.pdf

But we solved our problems by changing the device throughput limit that was initially set to 200 000 000 and dropped it to 50 000 000 this enabled us to use 7 1800-319U cameras. We also ran the  sudo sh -c 'echo 1024 > /sys/module/usbcore/parameters/usbfs_memory_mb' command as stated in the note.

Dropping down the throughput limit will change the frame rate, so consider using Bayer to enable more frame rate