astuff/avt_vimba_camera #114: Node crashes after failing to load nodelet to manager.

Eeaeau, 2023-02-16:
Hi! 

I was hoping someone here would have an idea of what could cause this error while trying either mono_camera launch files. I have tried both building this package locally and with the ROS package. 

I am using an Alvium U-052-C camera and have the vimba viewer working successfully.
 
Output:
<details>
<summary>Details</summary>

```
SUMMARY
========

PARAMETERS
 * /camera/acquisition_mode: Continuous
 * /camera/acquisition_rate: 20
 * /camera/balance_ratio_abs: 1.0
 * /camera/balance_ratio_selector: Red
 * /camera/binning_x: 1
 * /camera/binning_y: 1
 * /camera/camera_info_url: file:///opt/ros/m...
 * /camera/decimation_x: 1
 * /camera/decimation_y: 1
 * /camera/exposure: 50000
 * /camera/exposure_auto: Continuous
 * /camera/exposure_auto_alg: FitRange
 * /camera/exposure_auto_max: 50000
 * /camera/exposure_auto_min: 50000
 * /camera/exposure_auto_outliers: 50000
 * /camera/exposure_auto_rate: 50000
 * /camera/exposure_auto_target: 50000
 * /camera/exposure_auto_tol: 5
 * /camera/frame_id: camera
 * /camera/gain: 0
 * /camera/gain_auto: Continuous
 * /camera/gain_auto_adjust_tol: 5
 * /camera/gain_auto_max: 32
 * /camera/gain_auto_min: 0
 * /camera/gain_auto_outliers: 0
 * /camera/gain_auto_rate: 50
 * /camera/gain_auto_target: 50
 * /camera/guid: DEV_1AB22C02933D
 * /camera/height: 624
 * /camera/ip: 
 * /camera/iris_auto_target: 50
 * /camera/iris_mode: Continuous
 * /camera/iris_video_level_max: 110
 * /camera/iris_video_level_min: 110
 * /camera/line_mode: Input
 * /camera/line_selector: Line0
 * /camera/offset_x: 0
 * /camera/offset_y: 0
 * /camera/pixel_format: RGB8
 * /camera/ptp_mode: Off
 * /camera/show_debug_prints: True
 * /camera/stream_bytes_per_second: 45000000
 * /camera/sync_in_selector: SyncIn1
 * /camera/sync_out_polarity: Normal
 * /camera/sync_out_selector: SyncOut1
 * /camera/sync_out_source: GPO
 * /camera/trigger_activation: RisingEdge
 * /camera/trigger_delay: 0.0
 * /camera/trigger_mode: Off
 * /camera/trigger_selector: FrameStart
 * /camera/trigger_source: FixedRate
 * /camera/whitebalance_auto: Continuous
 * /camera/whitebalance_auto_rate: 100
 * /camera/whitebalance_auto_tol: 5
 * /camera/width: 816
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    avt_vimba_nodelet_manager (nodelet/nodelet)
    camera (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[avt_vimba_nodelet_manager-1]: started with pid [13381]
process[camera-2]: started with pid [13385]
[ INFO] [/camera] [1676543712.098514275]: Loading nodelet /camera of type avt_vimba_camera/MonoCameraNodelet to manager avt_vimba_nodelet_manager with the following remappings:
[ INFO] [/camera] [1676543712.100603561]: waitForService: Service [/avt_vimba_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [/avt_vimba_nodelet_manager] [1676543712.108418578]: Initializing nodelet with 8 worker threads.
[ INFO] [/camera] [1676543712.121326587]: waitForService: Service [/avt_vimba_nodelet_manager/load_nodelet] is now available.
terminate called after throwing an instance of 'AVT::System::Exception'
terminate called recursively
[FATAL] [/camera] [1676543714.169632500]: Failed to load nodelet '/camera` of type `avt_vimba_camera/MonoCameraNodelet` to manager `avt_vimba_nodelet_manager'
...
... shutting down processing monitor complete
done
```

</details>

Eeaeau, 2023-02-17:
We switched to Vimba 6.0 and now the above issue seems to be resolved.

However, we now get this error: 

`[ERROR] [1676656121.548986176]: [Vimba System]: Could not start Vimba system: TL not loaded.`

shervinkoushan, 2023-02-18:
For reference we are using Ubuntu 18.04, and it runs on an NVIDIA Jetson TX2 board

icolwell-as, 2023-02-22:
The `TL not loaded` error is common, and is usually something outside the control of the ROS driver. TL stands for "Transport Layer" if I remember correctly, and is something provided by the AVT VimbaCPP libs.
Do you get the TL error when running VimbaViewer with the new Vimba 6.0? or other example codes?

Here's another example of the TL error with some solutions:
https://github.com/astuff/avt_vimba_camera/issues/29

Eeaeau, 2023-02-22:
> The `TL not loaded` error is common, and is usually something outside the control of the ROS driver. TL stands for "Transport Layer" if I remember correctly, and is something provided by the AVT [...]

Thanks for your response! We want to deploy this on a drone without display, so we have not tested VimbaViewer. The examples do work though. Using the VimbaX driver does not seem to work. 

Currently, we suspect that our avt_vimba_camera ROS package is outdated and does not include USB support. We will update the issue when we have tested the latest version.

Eeaeau, 2023-02-23:
Updating to the 1.2.0 driver fixed our problem :)

closed 2023-02-23 by Eeaeau

icolwell-as, 2023-02-23:
ok awesome! Glad you have a working solution, thanks for following up.