astuff/avt_vimba_camera #115: Update Vimba version in ROS driver for compatibility with CSI2

lucleon, 2023-02-16:
The latest [Vimba for ARM](https://docs.alliedvision.com/Vimba_ReleaseNotes/ARM.html#summary) release includes CSI Transport Layer which is compatible with CSI-2 cameras such as 1800 C-507 c/m.

I updated the compiled VimbaC and VimbaCPP libraries to their new version (1.9.2 and 1.9.0) and was able to build the ROS Driver from source with catkin_make.

When starting the ROS node with 'roslaunch avt_vimba_camera mono_camera.launch guid:=DEV_00012C01D9C0' I'm getting a segmentation fault error. 
The backtrace with gdb looks like this:

_```
#0  0x0000fffff7780d20 in boost::filesystem::path::filename() const () from /lib/aarch64-linux-gnu/libboost_filesystem.so.1.71.0
#1  0x0000fffff7781d3c in boost::filesystem::path::extension() const () from /lib/aarch64-linux-gnu/libboost_filesystem.so.1.71.0
#2  0x0000fffff7781ee0 in boost::filesystem::path::replace_extension(boost::filesystem::path const&) () from /lib/aarch64-linux-gnu/libboost_filesystem.so.1.71.0
#3  0x0000fffff1cd5a34 in ?? () from /opt/Vimba_6_1/VimbaCSITL/CTI/arm_64bit/VimbaCSITL.cti
#4  0x0000fffff1cefca4 in ?? () from /opt/Vimba_6_1/VimbaCSITL/CTI/arm_64bit/VimbaCSITL.cti
#5  0x0000fffff1cf0154 in ?? () from /opt/Vimba_6_1/VimbaCSITL/CTI/arm_64bit/VimbaCSITL.cti
#6  0x0000fffff1d66134 in ?? () from /opt/Vimba_6_1/VimbaCSITL/CTI/arm_64bit/VimbaCSITL.cti
#7  0x0000fffff1d65cf8 in ?? () from /opt/Vimba_6_1/VimbaCSITL/CTI/arm_64bit/VimbaCSITL.cti
#8  0x0000fffff1d4c504 in GCInitLibEx () from /opt/Vimba_6_1/VimbaCSITL/CTI/arm_64bit/VimbaCSITL.cti
#9  0x0000fffff71f75c4 in ?? () from /home/user/avt_ws/src/avt_vimba_camera/lib/arm_64bit/libVimbaC.so
#10 0x0000fffff71fba3c in ?? () from /home/user/avt_ws/src/avt_vimba_camera/lib/arm_64bit/libVimbaC.so
#11 0x0000fffff71e7fac in VmbStartup () from /home/user/avt_ws/src/avt_vimba_camera/lib/arm_64bit/libVimbaC.so
#12 0x0000fffff7c00c9c in AVT::VmbAPI::VimbaSystem::Startup() () from /home/user/avt_ws/src/avt_vimba_camera/lib/arm_64bit/libVimbaCPP.so
#13 0x0000aaaaaab634b8 in avt_vimba_camera::AvtVimbaApi::start() ()
#14 0x0000aaaaaab4c3a8 in avt_vimba_camera::MonoCamera::MonoCamera(ros::NodeHandle&, ros::NodeHandle&) ()
#15 0x0000aaaaaab45504 in main ()
```_

To me this looks like an issue with the CSI transport layer, does anybody know if I'm missing some steps to adapt this ROS driver to the latest Vimba Version and to make it compatible with CSI TL?

The OS is Jetpack5 on a Jetson Xavier NX, ROS version is noetic, I was able to install the driver and run VimbaViewer and Vimba SDK Examples.

icolwell-as, 2023-02-22:
When you say "I updated the compiled VimbaC and VimbaCPP libraries to their new version (1.9.2 and 1.9.0)" do you mean you replaced the contents in the `lib` folder of this repo with the updated Vimba libraries?
Assuming yes, did you also update the header files located at `include/VimbaCPP` and `include/VimbaC`?

lucleon, 2023-02-22:
> When you say "I updated the compiled VimbaC and VimbaCPP libraries to their new version (1.9.2 and 1.9.0)" do you mean you replaced the contents in the `lib` folder of this repo with the updated [...]

Yes, I also updated the header files.
In the meantime I reached out to Alliedvision support, they said this is a known issue that can be fixed by either using libboost version 1.58 or using a fixed .cti file they provided me with.

closed 2023-02-22 by lucleon

sivaranjanija, 2023-06-12:
Hi @lucleon, I am also facing the same issue, and thanks for your comment, clarifies why the issue is. Would it be possible for you to attach the .cti file here, would be of great help. Thanks.

lucleon, 2023-06-12:
Can't upload it because the file type is not supported. I would suggest reaching out to the Alliedvision Support.

sivaranjanija, 2023-06-12:
sure, thanks