astuff/avt_vimba_camera #116: Adjusting the white balance

shervinkoushan, 2023-02-23:
Would it be possible to adjust the `balance_ratio` for both Red and Blue?

Currently only one at a time is supported, but the calibration we get from Vimba Viewer has values for both.

icolwell-as, 2023-02-23:
Are you using ROS1 or ROS2? 
The ROS2 driver has an improved parameter management system, so any feature/parameter available from the camera is also available as a ROS2 parameter.

If you are using ROS1, someone will need to implement the ROS parameter and open a PR.

Eeaeau, 2023-02-23:
> Are you using ROS1 or ROS2? The ROS2 driver has an improved parameter management system, so any feature/parameter available from the camera is also available as a ROS2 parameter.
> 
> If you are using ROS1, someone will need to implement the ROS parameter and open a PR.

We need to be using ROS1.

shervinkoushan, 2023-02-23:
Actually, it does not seem like setting even one value in the launch file works. 

For reference, we are using a 1800-U-052C USB camera and running on Ubuntu 18.0, branch melodic

icolwell-as, 2023-02-23:
What launch file are you using and with what args/parameters?
Maybe try the `ros1_master` branch, there were some unreleased improvements to the nodelet launch files, if you are using those.

Eeaeau, 2023-02-23:
> What launch file are you using and with what args/parameters? Maybe try the `ros1_master` branch, there were some unreleased improvements to the nodelet launch files, if you are using those.

We used the melodic branch. We have not used the nodelet as we had some issues with that. We will try ros1_master tomorrow :).

shervinkoushan, 2023-02-24:
We tried the `ros1_master` branch now. The `balance_ratio` parameters do not seem to have any effect here either. All our parameters are listed below:

<launch>
  
  <arg name="name" default="camera" doc="The name of the camera"/>
  <arg name="ip" default="" doc="The IP for the camera to connect to"/>
  <arg name="guid" default="DEV_1AB22C02933D" doc="The GUID for the camera to connect to"/>
  <arg name="frame_id" default="$(arg name)" doc="The frame id of the camera"/>
  <arg name="use_measurement_time" default="false" doc="Use the measurement time from the camera"/>
  <arg name="ptp_offset" default="-37" doc="Offset to add to the timestamp from the camera (s).
    Useful for converting from TAI to UTC time. Only applies if use_measurement_time is true"/>
  <arg name="camera_info_url" default="file://$(find avt_vimba_camera)/calibrations/calibration_example.yaml"/>
  <arg name="print_all_features" default="false" doc="Show all camera features on the console during startup"/>
  <arg name="image_proc" default="false" doc="Launch image_proc with driver"/>

  
  

  
  <arg name="acquisition_mode" default="Continuous"/>
  <arg name="acquisition_rate" default="10"/>
  
  <arg name="trigger_source" default="Off"/>
  <arg name="trigger_mode" default="Off"/>
  <arg name="trigger_selector" default="FrameStart"/>
  <arg name="trigger_activation" default="RisingEdge"/>
  <arg name="trigger_delay" default="0.0"/>
  
  <arg name="exposure" default="50000"/>
  <arg name="exposure_auto" default="Continuous"/>
  <arg name="exposure_auto_alg" default="FitRange"/>
  <arg name="exposure_auto_tol" default="5"/>
  <arg name="exposure_auto_max" default="50000"/>
  <arg name="exposure_auto_min" default="50000"/>
  <arg name="exposure_auto_outliers" default="50000"/>
  <arg name="exposure_auto_rate" default="50000"/>
  <arg name="exposure_auto_target" default="50000"/>
  
  <arg name="gain" default="10"/>
  <arg name="gain_auto" default="Off"/>
  <arg name="gain_auto_adjust_tol" default="5"/>
  <arg name="gain_auto_max" default="32"/>
  <arg name="gain_auto_min" default="0"/>
  <arg name="gain_auto_outliers" default="0"/>
  <arg name="gain_auto_rate" default="50"/>
  <arg name="gain_auto_target" default="50"/>
  
  <arg name="balance_ratio_abs" default="3"/>
  <arg name="balance_ratio_selector" default="Red"/>
  <arg name="whitebalance_auto" default="Off"/>
  <arg name="whitebalance_auto_tol" default="5"/>
  <arg name="whitebalance_auto_rate" default="100"/>
  
  <arg name="binning_x" default="1"/>
  <arg name="binning_y" default="1"/>
  <arg name="decimation_x" default="1"/>
  <arg name="decimation_y" default="1"/>
  
  <arg name="width" default="816"/>
  <arg name="height" default="624"/>
  <arg name="offset_x" default="0"/>
  <arg name="offset_y" default="0"/>
  
  <arg name="pixel_format" default="RGB8"/>
  
  <arg name="stream_bytes_per_second" default="45000000"/>
  
  <arg name="ptp_mode" default="Off"/>
  
  <arg name="sync_in_selector" default="SyncIn1"/>
[... 64 lines cut ...]
    <param name="width" value="$(arg width)"/>
    <param name="height" value="$(arg height)"/>
    <param name="offset_x" value="$(arg offset_x)"/>
    <param name="offset_y" value="$(arg offset_y)"/>

    <param name="pixel_format" value="$(arg pixel_format)"/>

    <param name="stream_bytes_per_second" value="$(arg stream_bytes_per_second)"/>

    <param name="ptp_mode" value="$(arg ptp_mode)"/>

    <param name="sync_in_selector" value="$(arg sync_in_selector)"/>
    <param name="sync_out_polarity" value="$(arg sync_out_polarity)"/>
    <param name="sync_out_selector" value="$(arg sync_out_selector)"/>
    <param name="sync_out_source" value="$(arg sync_out_source)"/>

    <param name="line_selector" value="$(arg line_selector)"/>
    <param name="line_mode" value="$(arg line_mode)"/>

    <param name="iris_auto_target" value="$(arg iris_auto_target)"/>
    <param name="iris_mode" value="$(arg iris_mode)"/>
    <param name="iris_video_level_min" value="$(arg iris_video_level_min)"/>
    <param name="iris_video_level_max" value="$(arg iris_video_level_max)"/>
  </node>
</launch>

Eeaeau, 2023-02-28:
Hmm, that could be the case! We do indeed see a lot of those messages. 

The naming in vimba viewer is different, should it match the ros driver config?
[image]

icolwell-as, 2023-03-01:
For the features supported by the Alvium 1800-U-052C, you'll want to review the feature reference document for Alvium:
https://cdn.alliedvision.com/fileadmin/content/documents/products/cameras/various/features/Alvium_Features_Reference.pdf

The feature reference document will tell you all the features and their exact names that are supported by alvium series cameras.
Then you can update the code here: https://github.com/astuff/avt_vimba_camera/blob/ros1_master/src/avt_vimba_camera.cpp#L1055
According to what parameters/features you want for your camera.

Another option is using the ROS2 driver along with the [ros1_bridge](https://github.com/ros2/ros1_bridge), since the ROS2 driver ensures all supported features are exposed as ROS2 params automatically.

shervinkoushan, 2023-03-01:
Thanks for pointing us in the right direction!

We got it working with this repository: https://github.com/Mkarasneh/avt_vimba_camera

Another question is how to achieve 12 bit output, but I can open another issue for that.

closed 2023-03-01 by shervinkoushan