astuff/avt_vimba_camera #126: Launch files broken for Iron

jmtatsch, 2023-10-16:
Hi,

I havent used Ros for quite some time so my knowledge may be a bit outdated, but I have trouble to launch the image producer with the provided launch files.

```
ros2 launch avt_vimba_camera Mako_G-319.launch.xml                                                                                                             
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [xml]: Unexpected attribute(s) found in `arg`: {'doc'}
```

Seems like the doc attribute has gone in the latest ros? But I couldn't find that in the ros docs...
Anyhow, after removing the doc attributes from the mako and the mono launch files, I seem to get further:

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mono_camera_exec-1]: process started with pid [2977458]
[mono_camera_exec-1] [INFO] [1697483760.607854395] [camera]: Parameters loaded
[mono_camera_exec-1] [INFO] [1697483760.636080665] [camera]: [Vimba System]: AVT Vimba System initialized successfully
[mono_camera_exec-1] [INFO] [1697483760.636130782] [camera]: Searching for cameras ...
[mono_camera_exec-1] [INFO] [1697483760.790366767] [camera]: Found camera named Alvium:
[mono_camera_exec-1] [INFO] [1697483760.790411354] [camera]:  - Model Name     : Alvium G1-319c
[mono_camera_exec-1] [INFO] [1697483760.790429630] [camera]:  - Camera ID      : DEV_000A47344617
[mono_camera_exec-1] [INFO] [1697483760.790436877] [camera]:  - Serial Number  : 04W3N
[mono_camera_exec-1] [INFO] [1697483760.790443348] [camera]:  - Interface ID   : enp0s31f6
[mono_camera_exec-1] [INFO] [1697483760.790449834] [camera]:  - Interface type : GigE
[mono_camera_exec-1] [INFO] [1697483760.790456495] [camera]:  - Access type    : Read and write access
[mono_camera_exec-1] [INFO] [1697483760.790985028] [camera]: Trying to open camera by ID: DEV_000A47344617
[mono_camera_exec-1] [INFO] [1697483761.269716863] [camera]: Opened connection to camera named Alvium with ID DEV_000A47344617
[mono_camera_exec-1] [WARN] [1697483763.270137553] [camera]: Feature DeviceTemperatureSelector: value unavailable
[mono_camera_exec-1] 	ERROR Invalid value.
[mono_camera_exec-1] [WARN] [1697483763.270242718] [camera]: Could not get feature GevTimestampTickFrequency
[mono_camera_exec-1] [INFO] [1697483763.291395650] [camera]: Ready to receive frames triggered by Software
[mono_camera_exec-1] [WARN] [1697483763.291477917] [camera]: Feature DeviceTemperatureSelector: value unavailable
[mono_camera_exec-1] 	ERROR Invalid value.
[mono_camera_exec-1] [INFO] [1697483763.311448209] [camera]: Configuring camera:
[mono_camera_exec-1] [INFO] [1697483763.318759096] [camera]:  - AcquisitionFrameRateEnable set to 0
[mono_camera_exec-1] [INFO] [1697483763.321025294] [camera]:  - AcquisitionMode set to Continuous
[mono_camera_exec-1] [INFO] [1697483763.321759654] [camera]:  - AcquisitionStatusSelector set to AcquisitionActive
[mono_camera_exec-1] [WARN] [1697483763.322812382] [camera]: Feature ActionDeviceKey is not readable.
[mono_camera_exec-1] [WARN] [1697483763.322828928] [camera]:  - Tried to set ActionDeviceKey to 0 but the camera used 94576465171584 instead
[mono_camera_exec-1] [INFO] [1697483763.324093401] [camera]:  - ActionGroupKey set to 0
[mono_camera_exec-1] [INFO] [1697483763.325345189] [camera]:  - ActionGroupMask set to 0
[mono_camera_exec-1] [INFO] [1697483763.325962466] [camera]:  - ActionSelector set to 0
[mono_camera_exec-1] [INFO] [1697483763.328936558] [camera]:  - AdaptiveNoiseSuppressionFactor set to 1
[mono_camera_exec-1] [INFO] [1697483763.333037060] [camera]:  - AutoModeRegionHeight set to 770
[mono_camera_exec-1] [INFO] [1697483763.337751924] [camera]:  - AutoModeRegionOffsetX set to 0
[mono_camera_exec-1] [INFO] [1697483763.341761388] [camera]:  - AutoModeRegionOffsetY set to 388
[mono_camera_exec-1] [INFO] [1697483763.342158493] [camera]:  - AutoModeRegionSelector set to AutoModeRegion1
[mono_camera_exec-1] [INFO] [1697483763.345729720] [camera]:  - AutoModeRegionWidth set to 2064
[mono_camera_exec-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
[mono_camera_exec-1]   what():  parameter 'feature/BalanceRatio' could not be set: Parameter {feature/BalanceRatio} doesn't comply with floating point range.
[ERROR] [mono_camera_exec-1]: process has died [pid 2977458, exit code -6, cmd '/opt/ros/iron/lib/avt_vimba_camera/mono_camera_exec --ros-args -r __node:=camera --params-file /tmp/launch_params_yqg2tsdd --params-file /tmp/launch_params_tjnf67y6 --params-file /tmp/launch_params_23wk6zge --params-file /tmp/launch_params_xbachyb4 --params-file /tmp/launch_params_hiin_b0l --params-file /tmp/launch_params_j_bu8keg --params-file /tmp/launch_params_hb9ktkbd --params-file /opt/ros/iron/share/avt_vimba_camera/config/Mako_G-319.yaml'].
```

But I guess that is a different issue.

jmtatsch, 2023-10-18:
Should I make a pull request with the doc attributes removed?

joshuasrcho, 2024-09-13:
Confirmed this is also an issue for humble. The solution is to change `doc` attribute to `description`. See [https://docs.ros.org/en/humble/How-To-Guides/Migrating-from-ROS1/Migrating-Launch-Files.html#arg](url).